jekhokie/raspberry-noaa-v2 #12: Twitter/Facebook Auto-Post of Captures

jekhokie, 2021-01-28:
Refactor the twitter publication and add the ability to post images directly to Facebook.

colinluthier, 2021-02-12:
I think this needs a filter of some sort. Perhaps like a post the best image of the day?

jekhokie, 2021-02-12:
I'd suggest if someone is taking this on to focus and start simple - have the primary image generated (the one that shows as the "face" of the capture in the "Captures" webpanel) be the image that is sent to twitter/facebook. Once the functionality is built, then start adding features based on things like "best", "selectable", etc.

I worry if you scope in filtering from the get-go, you might end up significantly complicating this to the point of it taking forever. :-)

colinluthier, 2021-02-12:
Some of the forks have each of these features and could be incorporated. V1 has the Twitter ability. I have seen a fork with the FB ability.

colinluthier, 2021-02-12:
Felix the Radio Guy also has a Discord script.

jekhokie, 2021-02-12:
Very cool - do you think you could get ahold of the discord script and post it here? I actually think I know someone who has the Facebook one, so I’ll grab that one.

jekhokie, 2021-02-12:
Functionality using IFTTT from Dom:
```
        # Send to email / facebook page: needs some filtration for bad images in due course
    if [ -f "${NOAA_OUTPUT}/images/${3}-122-rectified_ir.jpg" ]; then 
            mpack -s ${3}-${7}-"InfraRed" ${NOAA_OUTPUT}/images/${3}-122-rectified_ir.jpg trigger@applet.ifttt.com
    fi
    if [ -f "${NOAA_OUTPUT}/images/${3}-122-rectified.jpg" ]; then 
        mpack -s ${3}-${7} ${NOAA_OUTPUT}/images/${3}-122-rectified.jpg trigger@applet.ifttt.com
    fi
```

colinluthier, 2021-02-12:
# Discord
#exec(open("scripts/discord.py").read())

#direction = ""
#if altiwx.northbound:
#    direction = "Northbound"
#if altiwx.southbound:
#    direction = "Southbound"

#postToDiscord(altiwx.satellite_name + " " + altiwx.downlink_name, 65280, altiwx.elevation, direction, "AVHRR",  altiwx.date + " UTC", altiwx.frequency, output_file)

#altiwx.info("Webhook upload complete!")

colinluthier, 2021-02-12:
discord.py

```
from discord_webhook import DiscordWebhook, DiscordEmbed
import base64
import requests
import json
import subprocess
import os

webhook_url = "webhook" # Warning : Actual webhook in rtl-sdr discord!

# Upload an image to imgbb and return the URL
def upload_imagebb(image_path):
    with open(image_path, "rb") as file:
        url = "https://api.imgbb.com/1/upload"
        payload = {
            "key": "putYourApiKeyMilan",
            "image": base64.b64encode(file.read()),
        }
        res = requests.post(url, payload)
        data = json.loads(res.content)
        return data["data"]["url"]

def postToDiscord(title, color, elevation, direction, instrument, time, frequency, image):
    webhook = DiscordWebhook(url=webhook_url, username='Aang23 Station (bad antenna placement, WIP)', avatar_url='https://www.nesdis.noaa.gov/sites/default/files/assets/images/jpss.jpg')
    embed = DiscordEmbed(title=title, description="", color=color)
    embed.add_embed_field(name='Elevation', value=str(int(elevation)) + '°')
    embed.add_embed_field(name='Direction', value=direction)
    embed.add_embed_field(name='Instrument', value=instrument)
    embed.add_embed_field(name='Frequency', value=str(frequency / 1000000) + " Mhz")
    embed.add_embed_field(name='Time', value=time)
    goodimg = image + ".jpg"
    command = "convert -strip -interlace JPEG -quality 85% '" + image + "' '" + goodimg + "'"
    subprocess.Popen([command], shell=1).wait()
    embed.set_image(url=upload_imagebb(goodimg))
    os.remove(goodimg)
    webhook.add_embed(embed)
    response = webhook.execute()

```

jekhokie, 2021-02-17:
This is accomplished using the Email functionality and IFTTT - no further work necessary for Facebook in particular.

closed 2021-02-17 by jekhokie