jekhokie/raspberry-noaa-v2 #38: Support for Bias-T

Cystennin, 2021-02-09:
**Feature Request**
- Please add support for bias-T on NooElec NESDR SMArtee XR SDR
- If support for this is already present, please add documentation describing how to enable/disable the bias-T.

**Background**
One of the best combinations of SDR hardware for NOAA is using a NooElect SDR with Bias-T and a Sawbird+ LNA. When used with a high quality QFA antenna this combination can result in excellent coverage for passes over 25-30 deg. While there are options for using the Sawbird+ without bias-T support, such as using the external power supply, the bias-T is the most convenient and reduces the number of wires in a build and may also reduce QRM resulting from cheaper power supplies.

jekhokie, 2021-02-09:
Hi @KenStonerSDET - have you tried setting "enable_bias_tee" to "true" in the config/settings.yml file, and checked if that suits your need? For reference, setting it to true passes a "-T" switch to the "rtl_fm" command, so as long as the SDR you're using is compatible with rtl_fm and this option, I would suspect things to "just work".

Can you confirm?

KenStonerSDET, 2021-02-09:
I can confirm it seems to be working!

closed 2021-02-09 by KenStonerSDET