jekhokie/raspberry-noaa-v2 #41: Test Framework and CI for Future Changes

jekhokie, 2021-02-10:
As a means to ensure changes made don't break everything, and since sat passes don't occur regularly, it would be best to have a test framework in place with CI/CD to ensure future changes don't break the framework. One way to simulate/mock the satellite pass might be to use the APT encoding tool (https://github.com/gkbrk/apt-encoder) to generate an APT signal that the test framework can pick up and run through the various audio/image processors and check for outputs.

colinluthier, 2021-02-13:
If we skip the actual sound capture during testing and simply insert a known good prerecorded file we test everything but the radio.

jekhokie, 2021-03-12:
This is likely way harder now that we have such critical mass around things. We will likely be needing another major refactor soon to enhance the pluggable nature of this when we incorporate meteordemod - we’ll tackle this effort when we get there. For now, closing as stale.

closed 2021-03-12 by jekhokie