jekhokie/raspberry-noaa-v2 #46: Lock Admin Panel

thecivvie, 2021-02-10:
I know this maybe on your plans, but please add a password to the admin panel. I leave mine publicly facing [http://tcob1.tplinkdns.com]Renvyle(url)

jekhokie, 2021-02-10:
Thanks @thecivvie I'll leave this as an open feature requests, but just a note - I would actually be careful about exposing this (or the original) framework to the public internet. The way in which the webpanel is designed and the access it has to the file system on the Raspberry Pi have not been sufficiently tested for vulnerabilities (in my opinion), so at a minimum, I would recommend making sure your Raspberry Pi running this framework is off in its own HIGHLY restricted VLAN to prevent any access to other components on your network.

Again, this is not just a V2 thing - I would actually recommend the same for the original raspberry-noaa framework specifically.

thecivvie, 2021-02-10:
Yeah my 3 public facing Pi are behind a firewall and VLAN. But I hazard a guess, if someone wanted they would find a way

colinluthier, 2021-02-13:
This is not a feature I will use but a quick, and not rigorously secure method:
http://www.zubrag.com/scripts/password-protect.php

jekhokie, 2021-02-17:
As a note, this feature will almost certainly require a TLS-enabled site (even if just using self-generated certs). Adding that as a prerequisite to this.

Cirromulus, 2021-02-21:
I don't know about your setup, but my PI is behind a proxy pass on a "real" internet facing server.
It just does not pass /admin to the internet, making it only accessible from LAN.

Its a quick and dirty solution for now.

jekhokie, 2021-02-21:
@Cirromulus how are you configuring the directive in the web server - are you essentially placing an internal CIDR range on the `/admin` path (10.x, 192.x, etc) so that the web server only serves if the requestor IP matches the internal CIDR range?

Would you be willing to share your nginx config directive?

Cirromulus, 2021-02-21:
My setup is a bit more complicated, as accesses from the internet go to a separate server.
This server just acts as a proxy and does not forward the `/admin` path.
This way, the SSL-Certificate can be managed by the "internet-server" that manages some other sites as well.
Also, this way the PI is not connected directly to the internet, reducing the attack surface.

This is my apache config for this virtual host:
``` apache
<VirtualHost *:443>
    SSLEngine on
    ServerAdmin admin@[...].de
    ServerName subdomain.[...].de

    ProxyPass /admin !
    ProxyPass / "http://192.168.[...].147"
    ProxyPassReverse / "http://192.168.[...].147"

    Include /etc/letsencrypt/options-ssl-apache.conf
    SSLCertificateFile /etc/letsencrypt/[...]/.pem
    SSLCertificateKeyFile /etc/letsencrypt/[...]/.pem
</VirtualHost>
```

If you do all of it in one server, I suggest using a variant of [this](https://serverfault.com/questions/667798/allow-only-local-users-in-nginx/667863) SE-Post. Of course, you would have to fill in your local subnet, which may fail in the Ansible script, especially if a user has multiple interfaces.

Cirromulus, 2021-02-21:
I just fiddled around, this would work:

``` nginx
server {
  listen 80;
  root /var/www/wx-new/public/;
  index index.php index.html index.htm index.nginx-debian.html;
  autoindex on;

  server_name wx.home-new;

  location ~ /images/(.*) {
    alias /srv/images/$1;
  }

  location / {
    try_files $uri $uri/ /index.php;
  }

  location ~ \.php$ {
    include snippets/fastcgi-php.conf;
    fastcgi_pass unix:/var/run/php/php7.2-fpm.sock;
  }

  location /admin {
      try_files $uri $uri/ /index.php;
      allow 192.168.0.0/24;
      deny all;
  }

  # Rewrite the old path yyyy/mm/dd/ to /images/
  rewrite "^/image/\d{4}/\d{2}/\d{2}/(.+)$" /images/$1 permanent;
}
```

But beware, this is sensitive to the way your PI is connected to the internet.
If your router does not forward but instead does NAT, then for the PI it always seems that the router (thus, trusted LAN) accesses the server.

stevecrow74, 2021-02-27:
I just removed the line of script that removes the button for admin, if people can’t see it they can’t click it, if they really want to find a way they will regardless! And only those who have it installed will know what to add to the url to access the admin panel! Even though I might change the access url to something a bit more obscure! 
Which would be something to look into, the option of a personal access page for the admin panel and being able to hide the button for it.

closed 2021-03-02 by jekhokie