jekhokie/raspberry-noaa-v2 #48: Webinterface better look.

tomamdy, 2021-02-10:
When I delete a sat picture with X, I get the  picture Cesium wtih a lot of codes above it. May be it's possible to change?

jekhokie, 2021-02-10:
Hi @tomamdy - would you mind posting an image of what you're seeing? I'm wondering if it's complaining about permissions somehow, and if so, can you also post what version of the source code you have (the latest commit that shows in your git log locally)?

jekhokie, 2021-02-10:
@tomamdy can you send what version of the code you're using (the hash of the latest commit in your local cloned repo)?

tomamdy, 2021-02-11:
I am really a starter. I don't can find that.

jekhokie, 2021-02-11:
@tomamdy no worries - try this - do a "git pull" in the raspberry-noaa-v2 directory and then run the "./install_and_upgrade.sh" script and re-try the deletion to see if you still get the same error?

tomamdy, 2021-02-11:
And then this happens:

 git pull
remote: Enumerating objects: 279, done.
remote: Counting objects: 100% (250/250), done.
remote: Compressing objects: 100% (92/92), done.
remote: Total 177 (delta 97), reused 149 (delta 79), pack-reused 0
Receiving objects: 100% (177/177), 26.91 KiB | 491.00 KiB/s, done.
Resolving deltas: 100% (97/97), completed with 41 local objects.
From https://github.com/jekhokie/raspberry-noaa-v2
   7e3564f..aa8bdb3  master     -> origin/master
 * [new tag]         v1.0.0     -> v1.0.0
 * [new tag]         v1.1.0     -> v1.1.0
Updating 7e3564f..aa8bdb3
error: Your local changes to the following files would be overwritten by merge:
	ansible/roles/webserver/tasks/main.yml
	config/settings.yml.sample
Please commit your changes or stash them before you merge.
Aborting

jekhokie, 2021-02-11:
@tomamdy it looks like you possibly made changes to the following local files?

```
ansible/roles/webserver/tasks/main.yml
config/settings.yml.sample
```
I would suggest reverting these to the originally source-controlled files using the following commands:
```
git checkout -- ansible/roles/webserver/tasks/main.yml
git checkout -- config/settings.yml.sample
```
then re-run the following:
```
git pull
./install_and_upgrade.sh
```

tomamdy, 2021-02-11:
Yes, I did (may be) changed this file, config/settings.yml.sample but I thought it was only a sample.
I have done Your suggestions and can see the difference tonight (locall time).

tomamdy, 2021-02-12:
Nopasses, that has not given a good solution. I go install it again.

closed 2021-02-12 by tomamdy