jekhokie/raspberry-noaa-v2 #57: Request choice of Device through Device_index

Galoosis, 2021-02-11:
I would like to see the option to choose a device to connect to. I would like to be able to install this on a RPI with other programs that use dongles and be able to choose which dongle this program uses. IE: Device_index = 1 instead of it using the default device of zero.

jekhokie, 2021-02-11:
Note to self - this is likely just passing the ‘-d’ switch to the rtl_fm command (http://manpages.ubuntu.com/manpages/trusty/man1/rtl_fm.1.html). Update to include a device selector in config/settings.yml, default to 0, and provide instructions on how to get the device ID in case it’s unknown (rtl_eeprom or something similar that is already part of the rtl binaries installed).

jekhokie, 2021-02-11:
@Galoosis should be all set - do a git pull, add the new setting to your config/settings.yml file and specify your device ID that you want to use (the parameter is named "sdr_device_id" in that file - see the config/settings.yml.sample file for an example) and re-run ./install_and_upgrade.sh and you should be off to the races!

closed 2021-02-11 by jekhokie