jekhokie/raspberry-noaa-v2 #60: Save database to temp file before moving to main file

thecivvie, 2021-02-11:
Due to multiple power cuts today, my database got corrupted once or two. Maybe an option is to save the database to a temp file then copy the temp file as the main database file. I know this would mean copying the database to a temp file, writing the photos to the temp file then moving the temp file to main file

jekhokie, 2021-02-11:
@thecivvie Great idea! I think a little less aggressive way to handle this would be to do daily backups of the sqlite3 database, tagged with a `YYYYMMDD` extension for easy reference when restoring (something like `panel.db.20210211`, for example).

This seems fairly trivial - I'll see if I can't whip up a cron job that would do this on a 24-hour basis but that also manages to prune X versions of backups (so you don't end up with thousands of backed up database files).

jekhokie, 2021-02-13:
@thecivvie I've implemented nightly backups as a compromise on an "every write" approach, I hope this is ok with you. The issue I have with needing to do a copy on write is it's a LOT of churn on the SD card. Check out the README under the "Post Install" section - it links out to the `docs/db_backups.md` file that explains how to set it up and how to recover in case of failure.

closed 2021-02-13 by jekhokie