jekhokie/raspberry-noaa-v2 #67: Automated pass selection

colinluthier, 2021-02-12:
If passes are occurring simultaneously I would like the option to automatically pick:
Meteor First
Highest elevation NOAA Second

I really don’t want to wake up and have to select the passes each morning.

jekhokie, 2021-02-12:
@colinluthier just confirming - right now, the METEOR pass will always trump a NOAA pass (METEOR will cut off the rtl_fm recording when it starts if there is an active NOAA pass going on).

Given this, your ask is specifically that, for conflicting NOAA passes, it always chooses the highest elevation NOAA pass?

colinluthier, 2021-02-12:
Yes, exactly. I was thinking to add a priority script into the scheduling script.

jekhokie, 2021-02-12:
Got it, makes sense. Let me give this some thought around how we could approach this. We would still want things to show up in the pass list (it would show the little exclamation icon indicating overlap), but I'm thinking something like you mentioned in terms of priority, where the user could select "highest_elevation" or "first_available" as options, which the schedule script would then take into account. This is actually not likely to be trivial because you almost have to build the entire pass schedule first, then determine how to build out the jobs.

colinluthier, 2021-02-12:
Yes, my idea was to keep the schedule script the same and then prune it at the end.

jekhokie, 2021-02-12:
Open to a PR on this one if you have a good idea/approach - my only thought was we would need to get the 'at' job ID so that overlaps could be 'atrm' after being scheduled, but that's less than trivial given that the 'atq' interface doesn't have a great granular parsing mechanism. I suppose it's a pipe dream parser on the command line using awk/cut/whatever flavor of text manipulation you can come up with. :-)

colinluthier, 2021-02-12:
That’s a really good point. Perhaps the SQL database could record it as it was being scheduled?

jekhokie, 2021-02-12:
That was actually what I was hoping for - that I could capture the job ID of the AT job created upon creation, and then store the job ID in the database alongside the pass. That would make things MUCH simpler - but, of course, "at" is not so nice that way. :-)

jekhokie, 2021-02-12:
Well actually, here's one way using a lookahead grep:

```
x=$(echo "echo hello" | at $start_date 2>&1)
echo $x | grep -Po '(?<=job )[^ ]*'
```

That would output the job ID pretty consistently given the output of the "at" scheduling command. Then we could take the job ID and store it as an additional column in the predict_passes table. After all the scheduling is done, the very last thing in the higher-level `schedule.sh` script could parse all "active" passes ("1" in the "is_active" column) and for anything overlapping, apply the filter rule, and do a "update predict_passes set is_active = 0 where pass_start = <EPOCH_TIME>;"

colinluthier, 2021-02-12:
We are getting passed my abilities. I did find this code snippet:

Record the at queue state before scheduling, schedule your job, then find the additions:

$ date
Fri Mar  8 10:33:34 EST 2019
$ atq
3   2019-03-08 10:34 a bishop
$ atq > atq.1
$ echo "ls -l" > cmd.txt
$ at -f cmd.txt now + 2 min
job 4 at 2019-03-08 10:36
You have new mail in /var/spool/mail/bishop
$ atq > atq.2
$ comm atq.1 atq.2
3   2019-03-08 10:34 a bishop
    4   2019-03-08 10:36 a bishop
$ comm -23 atq.1 atq.2 | awk '{print $1}' # completed jobs
3
$ comm -13 atq.1 atq.2 | awk '{print $1}' # added jobs
4
As demonstrated, this is impervious to jobs finishing under you. Of course, if jobs are added in separate processes simultaneously, and you want to exclude those, a different solution would be called for (perhaps by grepping for the user submitting the job, or having different processes submit into separate at -q queues).

jekhokie, 2021-02-12:
Note to self, this should likely include selectable priorities like so when there are overlaps for NOAA:
- first_start: schedule the first sat in view (default today).
- last_start: schedule the last sat to arrive in view for capture.
- highest_elevation: schedule the satellite having the highest elevation of others.

Could get fancy later and apply noaa-priority as well (always schedule noaa18 if there is a conflict, etc) but I think the details above are enough to start and see where that takes us.

Note that for now, we’ll keep Meteor as always taking priority no matter what since the image quality is far better in almost all cases.

Doing all the scheduled captures and storing the ‘at’ job IDs in the passes database table seems like the right thing to do, and then doing a post-schedule ‘atrm’ for the jobs/captures that are filtered based on the filter priority set by the user.

colinluthier, 2021-02-12:
It might be best to have schedule.sh call prune_schedule.sh or something to keep the code separate. It matches the v2 style better.

jekhokie, 2021-02-12:
Agreed, good suggestion!

dom-robinson, 2021-05-29:
@colinluthier are you happy that the new 'admin' gui features address this manually, or still keen to automate it? I saw the issue is marked closed, but not sure it was solved...?

dom-robinson, 2021-10-09:
gonna assume this is ok and close..

closed 2021-10-09 by dom-robinson

colinluthier, 2021-10-09:
Good work!