jekhokie/raspberry-noaa-v2 #68: FTP upload

colinluthier, 2021-02-12:
It would be a big feature improvement to clone the local server, without the admin features, to a public server. This has been discussed on FB. wxtoimg had this feature but only in the GUI. I can easily write the code to do the FTP side nut the SQL side is beyond my experience level.

jekhokie, 2021-02-12:
@colinluthier just want to make sure I understand how this would work - is the idea that this would essentially just be a clone of the "Captures" page with the ability to view the captures? Removing admin makes sense to me, but I would suggest we should probably also leave "Passes" out of it since that is dependent on the SDR being present, no?

So, let me ask it this way - would it be sufficient to generate a static site from the "Captures" page and ship that as part of the FTP operation along with the corresponding images? (basically, just transferring all the captures to a remote server to be viewed on the public internet like a photo viewer site)?

thecivvie, 2021-02-12:
excellent idea of generating the html and uploading it along with the images

jekhokie, 2021-02-12:
Could someone walk me through the actual pattern for this? There are obviously many different hosting sites and different structures for uploading files. Would it be sufficient to create a *.tgz (compressed) file that contained all of the image files in `/srv/images` and content in `/var/www/wx-new` and then have the file FTP'd to the remote site and the user would un-pack them manually? Or does the remote hosting typically have a directory for images and directory for web content that the files should be FTP'd to directly? Or something else?

thecivvie, 2021-02-12:
I would say the easiest for you is to generate either .php or .html and the images in the same directory and then we can script an upload to our own websites. That way we could clear the existing files out once a new upload is required

jekhokie, 2021-02-12:
Note to self - this is going to be HIGHLY complicated to try and generate static content from the existing site components based on the way pagination works with the SQL database, etc. It's likely preferred that a static content generator would do something like the following:

1. Generate a static HTML page with static pagination (https://pagination.js.org/ or similar) built from the database records for `decoded_passes` and static links to local image files (transpose the `/srv/images` directory to `./assets/images` for local reference).
2. Copying of image files to local `./assets/images` directory that is referred to by the static page image source links.

Item 2 is likely not a "copy" on the local Pi but, rather, construction of a `.tgz` file that aligns the source images within and would explode them into the subdirectory `./assets/images` when unpacked.

@thecivvie and @colinluthier - is it reasonable to expect that this generator would produce a compressed `.tgz` file that contains the static site and respective images, or are you looking to have it do something like run an FTP command to transfer the static file and associated images directly to a remote source?

colinluthier, 2021-02-12:
wxtoimg just sends all the raw data. It has some code that generates a page and uploads a new index.html every time it uploads an image. I modified mine a fair amount but the image display is unaltered.

<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
  | <html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">
  | <head>
  | <meta http-equiv="content-type" content="text/html;charset=iso-8859-1" />
  | <meta name="expires" content="Fri, 08 Jan 2021 02:56:48 GMT" />
  | <meta http-equiv="imagetoolbar" content="no" />
  | <meta name="keywords" content="weather, satellite, images, Napa, United States, WXtoImg" />
  | <meta property="og:type" content="website" />
  | <meta property="og:url" content="http://designerinlight.com/wxdata/" />
  | <meta property="og:title" content="Weather satellite images recorded at Napa, United States." />
  | <meta property="og:description" content="Latitude: 38.296, Longitude: -122.307" />
  | <meta property="og:image" content="http://designerinlight.com/wxdata/preview.jpg" />
  | <title>Weather Satellite Images for Napa, United States</title>
  | <style type="text/css">
  | 
  | </style>
  | </head>
  | <body style="background-color:black;">
  | <div id="wx">
  | <div class="wxflag">
  | <h1>Weather satellite images recorded at Napa, United States.</h1>
  | <h4>Latitude: 38.296, Longitude: -122.307</h4>
  | <h4>Receiver: RTL-SDR Blog v3</h4>
  | <h4>Antenna: QFH</h4>
  | <h4>LNA: SawBird NOAA</h4>
  | <h4><pre> Software: WXtoImg version 2.11.2 beta
  | Orbitron 3.71
  | SDR# v1.0.0.1732
  | Tracking DDE CLient v1.2
  | VBCABLE Driver Pack43</pre></h4>
  | <p>
  | Click on any image below for a full sized image.
  | Times are shown in Pacific Standard Time.
  | </p>
  | <noscript>
  | <p id="wxwarning">
  | You must enable javascript in your web browser to see all the images and more detailed information.
  | </p>
  | </noscript>
  | </div>
  | <p />
  | <script type="text/javascript">
  | /* <![CDATA[ */
  | var xlang = new Array();
  |  
  | // -----------------------------------------------------
  | // language transation for satellite info here.
  | // change only right-hand side, it must be surrounded
  | // by double quotes and end with a semicolon.
  | // You can add other projection translations as needed.
  | // You can also translate other enhancement names.
  | xlang['click']		= "click image for full-size image";
  | xlang['Composite Image']= "Composite Image";
  | xlang['on']		= "on";
  | xlang['at']		= "at";
  | xlang['southbound']	= "Southbound";
  | xlang['northbound']	= "Northbound";
  | xlang['E']		= "E";		// Abbrev. for East
  | xlang['W']		= "W";		// Abbrev. for West
  |  
  | xlang['Pass Start']	= "Pass Start";
  | xlang['Last Pass']	= "Last Pass";
  | xlang['Enhancement']	= "Enhancement";
  | xlang['Projection']	= "Projection";
  | xlang['Direction']	= "Direction";
  | xlang['Max Elevation']	= "Max Elevation";
  | xlang['Solar Elevation']= "Solar Elevation";
  | xlang['Pass Duration']	= "Pass Duration";
[... 426 lines cut ...]
  | var dots = document.getElementsByClassName("dot");
  | for (i = 0; i < slides.length; i++) {
  | slides[i].style.display = "none";
  | }
  | slideIndex++;
  | if (slideIndex > slides.length) {slideIndex = 1}
  | for (i = 0; i < dots.length; i++) {
  | dots[i].className = dots[i].className.replace(" active", "");
  | }
  | slides[slideIndex-1].style.display = "block";
  | dots[slideIndex-1].className += " active";
  | setTimeout(showSlides, 1000); // Change image every 1 seconds
  | }
  | </script>
  |  
  | <iframe  width="500" height="600" src="https://forecast.weather.gov/product.php?site=NWS&issuedby=MTR&product=AFD&format=txt&version=1&glossary=0" frameborder="0" scrolling="auto" allowfullscreen>
  | </iframe>
  |  
  |  
  |  
  | </body>
  | </html>
 

``

colinluthier, 2021-02-12:
You can see the output of wxtoimg at designerinlight/wxdata. It makes no effort to compress the uploads. The index page is generated each time an pass ins completed. I changed a few things and added some weather things at the bottom. He has some premade templates that you can copy and change to suit your site.

jekhokie, 2021-02-12:
Hey Colin - do you have a direct link to the example? I got a 404 when I tried to visit your site wxdata page.

colinluthier, 2021-02-12:
```

<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
--
  | <html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">
  | <head>
  | <meta http-equiv="content-type" content="text/html;charset=iso-8859-1" />
  | <meta name="expires" content="Fri, 08 Jan 2021 02:56:48 GMT" />
  | <meta http-equiv="imagetoolbar" content="no" />
  | <meta name="keywords" content="weather, satellite, images, Napa, United States, WXtoImg" />
  | <meta property="og:type" content="website" />
  | <meta property="og:url" content="http://designerinlight.com/wxdata/" />
  | <meta property="og:title" content="Weather satellite images recorded at Napa, United States." />
  | <meta property="og:description" content="Latitude: 38.296, Longitude: -122.307" />
  | <meta property="og:image" content="http://designerinlight.com/wxdata/preview.jpg" />
  | <title>Weather Satellite Images for Napa, United States</title>
  | <style type="text/css">
  | <!--
  | #wx {
  | font-family: sans-serif, Helvetica, Arial, Geneva, Swiss, SunSans-Regular;
  | font-size: 88%;
  | color: #FFFFFF;
  | margin-left: 30px;
  | margin-right: 5px;
  | }
  | #wxwarning {
  | color: #e33;
  | font-weight: bold;
  | }
  |  
  | .wxflag {
  | width: 90%;
  | background-color: #000000;
  | padding: 10px 10px;
  | font-weight: bold;
  | margin-left: 10px;
  | }
  | .wxflag h1 {
  | color: #FFFFFF;
  | font-size: 171%;
  | padding-bottom: 3px;
  | margin-top: 0px;
  | margin-bottom: 0px;
  | }
  | .wxflag h2 {
  | color: #FFFFFF;
  | font-size: 143%;
  | padding-bottom: 3px;
  | margin-bottom: 0px;
  | }
  | .wxflag h3 {
  | color: #FFFFFF;
  | font-size: 124%;
  | padding-top: 2px;
  | margin-top: 0px;
  | }
  | .wxflag h4 {
  | color: #FFFFFF;
  | font-size: 100%;
  | padding-top: 2px;
  | margin-top: 0px;
  | padding-bottom: 2px;
  | margin-bottom: 0px;
  | }
  |  
  | .wximage {
  | display: block;
  | float: left;
  | background-color: #000000;
  | padding: 10px 10px;
[... 584 lines cut ...]
  | var slides = document.getElementsByClassName("mySlides");
  | var dots = document.getElementsByClassName("dot");
  | for (i = 0; i < slides.length; i++) {
  | slides[i].style.display = "none";
  | }
  | slideIndex++;
  | if (slideIndex > slides.length) {slideIndex = 1}
  | for (i = 0; i < dots.length; i++) {
  | dots[i].className = dots[i].className.replace(" active", "");
  | }
  | slides[slideIndex-1].style.display = "block";
  | dots[slideIndex-1].className += " active";
  | setTimeout(showSlides, 1000); // Change image every 1 seconds
  | }
  | </script>
  |  
  | <iframe  width="500" height="600" src="https://forecast.weather.gov/product.php?site=NWS&issuedby=MTR&product=AFD&format=txt&version=1&glossary=0" frameborder="0" scrolling="auto" allowfullscreen>
  | </iframe>
  |  
  |  
  |  
  | </body>
  | </html>

```

nomadixmedia, 2021-05-16:
An easy way to do this is to run node-red and watch the image folder, get it to create the HTML and upload it to a webserver.

dom-robinson, 2021-09-29:
I have put this on my to-do queue. Plan is to simply expose an 'ftp' option in settings.yml and to let the user create thier own ```curl``` command (with an example in the settings.yml comments) so they can work the ftp process in, but deal with all thier own nuances of security and ftp versions. Any other method will open RN2 up to a significant amoutn of engineering to manage all the PASV / binary / port settings - something that will be unique to each user. curl supports ftp well but the user will need to take a little responsibility to work out the setup they need for thier own workflow...