jekhokie/raspberry-noaa-v2 #74: bias t support with different rtl_fm [Error]

dom-robinson, 2021-02-12:
**Describe the bug**
When using Bias T settings with rtl_fm in the noaa record scripts it crashes if set to -T and using the wrong version of rtl_fm. Not all rtl_fm versions support Bias T

Better info here: https://www.rtl-sdr.com/forum/viewtopic.php?t=1887

jekhokie, 2021-02-12:
This likely belongs as a note in the `docs/troubleshooting.md` documentation for users.

jekhokie, 2021-02-12:
Note to self, when this goes into the docs, it likely needs to call out that the version of `rtl_fm` installed with this framework actually contains this functionality working as expected, but if people are using non-Raspberry Pi OS native images that come pre-packaged with `rtl_fm`, it's possible those versions do not have or support the Bias-Tee functionality.

colinluthier, 2021-02-12:
There is a workaround where you enable the Bias Tee in the RTL-SDR firmware and then you can plug it into any rlt_fm version.

closed 2021-02-13 by jekhokie