jekhokie/raspberry-noaa-v2 #88: Bias Tee

dom-robinson, 2021-02-13:
Various issues with BIAS TEE - this is a note from Colin / JediBeer's tips that Justin wanted the inssues:

--snip_
I need to sort out what I broke last night. I did re flash my eprom to enable software control of the bias tee again. I think that after the pass we need to send the command to turn off the bias T. At least with my old Window’s workflow it would stay on and I would like to lower the pi power consumption.

jekhokieToday at 2:11 PM
So I never really figured out how to turn it off to be honest - mine stays on continuously as well. If you figure it out I’m all ears and would love a pull request showing how you did it. :slight_smile:
If it’s some kind of rtl_eeprom command sequence we can append that to the receive scripts for sure
I honestly wish rtl_fm would handle turning it off when it terminates given that binary is actually enabling it with the ‘-T’ command in the first place

JediBeerToday at 2:17 PM

git clone https://github.com/rtlsdrblog/rtl-sdr-blog
cd rtl-sdr-blog
mkdir build
cd build
cmake .. -DDETACH_KERNEL_DRIVER=ON
make
cd src
./rtl_biast -b 1
GitHub
rtlsdrblog/rtl-sdr-blog

Modified Osmocom drivers with enhancements for RTL-SDR Blog V3 units. - rtlsdrblog/rtl-sdr-blog

If have this branch of the driver you send
 ./rtl_bias -b 0
If you use the code to turn it on rtl_fm will turn it off unless you have the -T switch.
I need to make a fork and learn how GitHub works to add changes. I saw the little tutorial you wrote on it and it is making more sense.

colinluthier, 2021-02-13:
RTL-SDR V3 Only:

Forcing the Bias Tee to be Always On

If you are using our RTL-SDR-Blog driver branch you can force the bias tee to be always on by setting a flag in the EEPROM. The rtl_eeprom command is "rtl_eeprom -b y". Run the opposite command "rtl_eeprom -b n" to disable the forced bias tee.

This might be useful in the troubleshooting section.

colinluthier, 2021-02-14:
Dom, In your quote from my Discord: ./rtl_bias -b 0

Needs to be: ./rtl_biast -b 0

jekhokie, 2021-02-20:
Closing this for now - if we run into future issues with bias-tee settings we can re-visit this.

closed 2021-02-20 by jekhokie

colinluthier, 2021-02-20:
I am not convinced we have this shaken out yet. I think we need to do version checks on preinstalled sdr drivers and control software. It’s not a problem on a fresh install but we can’t assume we are not installing over another install.