jekhokie/raspberry-noaa-v2 #91: Create Test Script for Local Testing

jekhokie, 2021-02-13:
Create a test script that would trigger the receive scripts to record audio for a predetermined length of time and perform processing/update the database (including predict_passes) such that the webpanel would show the result (even if a non-useful image) for local troubleshooting/testing.

If we get fancy, we could also break apart the receive scripts and allow passing in of a bitstream, wav, or qpsk file that the receive scripts would go through processing for and ignore just the radio receive part. Lots of options here.

colinluthier, 2021-02-14:
I have been thinking if this a lot. We could have two wave files. A good grab from NOAA and a good grab from Meteor. Then the receiving files would look for a test flag and substitute the rtl_fm call with the wav file.

jekhokie, 2021-02-14:
Agree - I usually end up commenting out the rtl_fm or gnuradio capture lines in the scripts and running the image through that way, but there are also database considerations to be had. You're def down the right track though.

colinluthier, 2021-03-05:
Interesting, today I really need a script that pretends its a sdr dongle.

jekhokie, 2021-03-12:
Same issue here as in #41 - stale ticket, and not likely possible without ripping everything apart (refactor). We can open a new request when we get there.

closed 2021-03-12 by jekhokie