jekhokie/raspberry-noaa-v2 #93: VNC Capability with GQRX and OpenWebRX

jekhokie, 2021-02-13:
It would be useful to have VNC enabled (as an option) and a few tools such as GQRX and OpenWebRX installed on the OS for remote GUI-based SDR functionality. See if these can be incorporated as an optional "install_gui_tools" and "remote_management" config options where Ansible would install and configure GQRX and OpenWebRX, with VNC installed/configured, respectively.

Note: Keeping the "remote_management" option separate from the software install would provide the ability to enable/disable (turn on/off) the VNC functionality for remote administration in case people want to turn it off or on for whatever reason (even after GUI tools are installed).

This *may* require installing a ton of other packages related to an X windows environment, bloating the image, so we should be clear about what this ends up installing and a note that the additional processing required for the window environment *may* have an impact on Pi instances that are lower in specs.

colinluthier, 2021-02-14:
I have the normal raspian desktop with xscreen on mine. I keep VNC enabled but rarely need it.

jekhokie, 2021-03-04:
Closing in favor of #220 - the other feature will implement a "streaming" capability to enable plugging in any number (and format) of plugins and tooling, which will be much more robust and flexible/scalable.

closed 2021-03-04 by jekhokie