jekhokie/raspberry-noaa-v2 #94: Personalisation

dom-robinson, 2021-02-13:
Be nice to have a secondary header (if you see what i mean) into which we can drop our own logo and header welcome / website etc.

jekhokie, 2021-02-13:
Are you referring to the webpanel specifically, or on the images? If on the webpanel, where would you like the logo specifically/can you articulate how you'd like to see it?

colinluthier, 2021-02-13:
I am working on a couple more niceties. Some more control on the maps and the ability to add the ground station to the image. I have made a fork to test.

jekhokie, 2021-02-14:
@colinluthier you can already add the ground station using the `noaa_map_crosshair_enable` setting in `config/settings.yml` if that's what you're referring to?

colinluthier, 2021-02-14:
No I was hoping to add Napa, Ca to the annotation.

colinluthier, 2021-02-14:
Or some radio guys would add their station ID. The more I think of this the annotation might need to be its own code. Font size and multi oil lines with static text would be a big step towards personalization.

colinluthier, 2021-02-14:
It’s almost like we need an html header that is editable and positionable and not changed by pulls.

jekhokie, 2021-02-14:
I think this is a great idea - maybe I’ll stick an example in a file like `config/annotation.sample` that users can copy to their own annotation file in the config directory, use variable placeholders in the template that will be substituted with the vars from the capture. That way the annotation file persists upgrades and is completely user-controlled.

colinluthier, 2021-02-14:
The comments would be the instructions. I like it.

mbruski, 2021-02-14:
I would also like to add the maidenhead grid square for my station to the image annotation as well.

colinluthier, 2021-02-16:
Adding a logo is pretty straightforward. I am going to add some documentation here. 
https://legacy.imagemagick.org/Usage/annotating/#anno_on

jekhokie, 2021-02-16:
I figured out a decent way to approach this - the idea would be to have an HTML template file in Jinja2 template language that exists as something like `config/annotation.html.j2` with some sane defaults. It will also have a list of variables that can be used, wrapped in double curly braces like this: `{{ sat_name }}` where users want to have fields dynamically populated. The benefit here is people can simply create HTML to their hearts desires with variables that we can expand on later as well, and there will then be 2 separate processing abilities:

1. A script that can render an image of the template so people can iterate on their template and see what the output would be.
2. Functionality that will convert the html to an image (basically what item 1 does above) and inlay it in the position where the annotation belongs on the images.

Procedurally, I tested a way to do this - which includes the following using the `wkhtmltox` package:
```
# create an image out of the html file
wkhtmltoimage --format png --quality 100 --transparent file:///some-file.html /tmp/out.png
# create a transparent background for the image so it doesn't disrupt the capture image when inlaid
convert -format png /tmp/out.png -background none /tmp/out2.png
```

The trick is before the first step, some custom code would need to render the actual HTML from the template with the variables injected (fairly trivial to do using Python).

This approach would have the benefit of being able to create an annotation with almost anything that HTML supports (styles, DOM elements, images, etc.) with the exception of (of course) clickable elements like links.

I'm pretty confident I have a way to get this working for a first cut.

closed 2021-02-25 by jekhokie