jekhokie/raspberry-noaa-v2 #96: [Error] No Images

colinluthier, 2021-02-13:
Since loading the latest build last night I have not gotten images. I thought I may have changed a file so I did a fresh pull this morning and I still didn't get an image. I am going to try this evenings passes with bias t = false.

colinluthier, 2021-02-13:
```
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 38.60 dB.
activated bias-T on GPIO PIN 0
Allocating 15 zero-copy buffers
Tuned to 138392500 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 60000 Hz.
Signal caught, exiting!

User cancel, exiting...
13-02-2021 09:42 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Producing spectrogram
13-02-2021 09:42 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Bulding pass map
Satellite: NOAA 18
Pass Start: 2021-02-13 17:27:22 UTC
Pass Duration: 15:14
Elevation: 37
Azimuth: 98
Direction: southbound
..13-02-2021 09:47 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
wxtoimg: warning: Narrow IF bandwidth (set Recording Options/receiver type), low S/N, or volume too high
Gain: 48.8
```

colinluthier, 2021-02-13:
The Meteor scripts also need a little correction.

colinluthier, 2021-02-13:
Ah, when ~/.noaa-v2.conf is written Receiver_Gain becomes Gain. Hmmm....

jekhokie, 2021-02-13:
Are you seeing images in /srv/images/ at all? Also are you using rtl_fm or the gnuradio version?

What does your /home/pi/.noaa-v2.conf file look like?

colinluthier, 2021-02-13:
Yes, I am getting an image that has a map but it is all noise.
```
---
# base station configurations
#   latitude: south values are negative
#   longitude: west values are negative
latitude: 38.2
longitude: -122.3

# time zone offset from UTC (for example, '-5' for US Eastern)
timezone_offset: -8

# receiver settings
#   meteor_receiver - which receiver method to use (either 'rtl_fm' or 'gnuradio')
#     **WARNING**: 'gnuradio' does not work with certain SDR devices (e.g. it will not currently
#                  work with a RTL-SDR v3 dongle, as no image will be decoded from the bitstream)
#   sdr_device_id - which SDR device to use (use rtl_eeprom to see available devices)
#   enable_bias_tee - driving power to LNA, for example
#   receiver_gain - receiver gain
#   receiver_freq_offset - receiver frequency offset (PPM)
meteor_receiver: 'rtl_fm'
sdr_device_id: 0
enable_bias_tee: false
#Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
receiver_gain: 38.6
receiver_freq_offset: 0

# whether to schedule specific orbiting objects for capture
schedule_noaa: true
schedule_meteor: true

# whether audio files should be deleted after images are created
delete_audio: true

# processing settings
#   flip_meteor_image - whether the meteor image should be flipped
#   produce_spectrogram - whether to produce a spectrogram image of the audio recording
#   noaa_crop_telemetry - whether to crop the left/right telemetry in image captures
#   image_annotation_location - where to place the annotation in images - valid options are:
#        NorthWest, North, NorthEast, West, Center, East, SouthWest, South, SouthEast
flip_meteor_image: true
produce_spectrogram: true
noaa_crop_telemetry: true
image_annotation_location: 'SouthWest'

# thresholds for scheduling captures - enables avoiding an attempt
# to capture imagery if objects are lower than these degree elevation thresholds
sat_min_elevation: 30
sun_min_elevation: 10

# noaa map configurations
# http://usradioguy.com/wp-content/uploads/2020/05/wxtoimgcommand-line.pdf
#
# note - colors are in format 0xRRGGBB (only applicable when the
#        feature is enabled), where:
#   RR: Red hex value
#   GG: Green hex value
#   BB: Blue hex value
# Colors can alternatively be specified as one of the following:
#   black, white, gray, light-gray, dark-gray, red, pink dark-red,
#   light-red, # green, light-green, dark-green, black-green, blue,
#   light-blue, dark-blue, black-blue, yellow, light-yellow, dark-yellow,
#   magenta, light-magenta, dark-magenta, cyan, light-cyan, dark-cyan,
#   orange, dark-orange, purple, lavender, violet, navy, turquoise,
#   aquamarine, chartreuse, gold, beige, tan, brown, and maroon
#
#   noaa_map_crosshair_enable - whether to place a crosshairs on the base station location
#   noaa_map_crosshair_color - color of base station crosshair
#   noaa_map_grid_degrees - latitude/longitude lines drawn every grid degrees (default 10.0, 0.0 to disable)
#   noaa_map_grid_color - color of gridlines for latitude/longitude
noaa_map_crosshair_enable: true
noaa_map_crosshair_color: "0xffff00"
noaa_map_grid_degrees: 10.0
noaa_map_grid_color: "0xff0000"

# locale settings for timezone and language
#   timezone: see https://www.php.net/manual/en/timezones.php
#   lang_setting: see the 'webpanel/App/Lang' folder for available
#                 languages (2-letter filename - e.g. ar, de, en, es, sr)
timezone: America/Los_Angeles
lang_setting: en

# port to run the webpanel on
web_port: 80

# log level for output from scripts
log_level: DEBUG

# whether to enable the satvis visualization for satellite tracking
# in the passes view - note that this iframe-driven visualization is
# by default disabled on "extra-small" devices such as phones due to
# the processing and space requirements
enable_satvis: false

# pruning capabilities - must be configured in cron (see documentation)
#   delete_oldest_n - how many oldest captures to delete on each run
#   delete_older_than_n - delete all images older than this many days
delete_oldest_n: 0
delete_older_than_n: 14
...
# © 2021 GitHub, Inc.

```

jekhokie, 2021-02-13:
Hm - have you had a working version of the code yet? When was the last version cloned that worked for you? I can check the changes since then to see what stands out?

jekhokie, 2021-02-13:
Another thought here - I saw you have bias tee disabled - are you sure the LNA (if you’re using one) is actually powered when the pass recording is happening?

colinluthier, 2021-02-13:
The light has not turned off but I enabled it again in the eeprom before turning it off. The first release of 2.1.1 was working well. I really think its a gain issue. 

but I am not sure, the log has two entries for gain:

```
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 38.60 dB.
```

```
wxtoimg: warning: Narrow IF bandwidth (set Recording Options/receiver type), low S/N, or volume too high
Gain: 48.8
```

colinluthier, 2021-02-13:
The images captured have mostly noise but the map aligns.

jekhokie, 2021-02-13:
Alright I’ll test/troubleshoot the gain setting tonight and let you know if I find anything. Thanks.

rto54, 2021-02-15:
Hello,
Since the last update. I have no more images going back.

---
# base station configurations
#   latitude: south values are negative
#   longitude: west values are negative
latitude: 48.625117
longitude: 6.297899

# time zone offset from UTC (for example, '-5' for US Eastern)
timezone_offset: +1

# receiver settings
#   meteor_receiver - which receiver method to use (either 'rtl_fm' or 'gnuradio')
#     **WARNING**: 'gnuradio' does not work with certain SDR devices (e.g. it will not currently
#                  work with a RTL-SDR v3 dongle, as no image will be decoded from the bitstream)
#   sdr_device_id - which SDR device to use (use rtl_eeprom to see available devices)
#   enable_bias_tee - driving power to LNA, for example
#   receiver_gain - receiver gain
#   receiver_freq_offset - receiver frequency offset (PPM)
meteor_receiver: 'rtl_fm'
sdr_device_id: 0
enable_bias_tee: true
receiver_gain: 50
receiver_freq_offset: -6

# whether to schedule specific orbiting objects for capture
schedule_noaa: true
schedule_meteor: true

# whether audio files should be deleted after images are created
delete_audio: false

# processing settings
#   flip_meteor_image - whether the meteor image should be flipped
#   produce_spectrogram - whether to produce a spectrogram image of the audio recording
#   noaa_crop_telemetry - whether to crop the left/right telemetry in image captures
#   image_annotation_location - where to place the annotation in images - valid options are:
#        NorthWest, North, NorthEast, West, Center, East, SouthWest, South, SouthEast
#   produce_noaa_pristine_image - whether to produce a pristine image (unmodified) for larger
#        composite-based use cases
flip_meteor_image: true
produce_spectrogram: true
noaa_crop_telemetry: true
image_annotation_location: 'NorthEast'
produce_noaa_pristine_image: true

# thresholds for scheduling captures - enables avoiding an attempt
# to capture imagery if objects are lower than these degree elevation thresholds
sat_min_elevation: 30
sun_min_elevation: 10

# noaa map configurations
# http://usradioguy.com/wp-content/uploads/2020/05/wxtoimgcommand-line.pdf
#
# note - colors are in format 0xRRGGBB (only applicable when the
#        feature is enabled), where:
#   RR: Red hex value
#   GG: Green hex value
#   BB: Blue hex value
# Colors can alternatively be specified as one of the following:
#   black, white, gray, light-gray, dark-gray, red, pink dark-red,
#   light-red, # green, light-green, dark-green, black-green, blue,
#   light-blue, dark-blue, black-blue, yellow, light-yellow, dark-yellow,
#   magenta, light-magenta, dark-magenta, cyan, light-cyan, dark-cyan,
#   orange, dark-orange, purple, lavender, violet, navy, turquoise,
#   aquamarine, chartreuse, gold, beige, tan, brown, and maroon
#
#   noaa_map_crosshair_enable - whether to place a crosshairs on the base station location
#   noaa_map_crosshair_color - color of base station crosshair
#   noaa_map_grid_degrees - latitude/longitude lines drawn every grid degrees (default 10.0, 0.0 to disable)
#   noaa_map_grid_color - color of gridlines for latitude/longitude
noaa_map_crosshair_enable: true
noaa_map_crosshair_color: "0xffff00"
noaa_map_grid_degrees: 10.0
noaa_map_grid_color: "0xff0000"

# locale settings for timezone and language
#   timezone: see https://www.php.net/manual/en/timezones.php
#   lang_setting: see the 'webpanel/App/Lang' folder for available
#                 languages (2-letter filename - e.g. ar, de, en, es, sr)
timezone: Europe/Paris
lang_setting: en

# port to run the webpanel on
web_port: 80

# log level for output from scripts
log_level: DEBUG

# whether to enable the satvis visualization for satellite tracking
# in the passes view - note that this iframe-driven visualization is
# by default disabled on "extra-small" devices such as phones due to
# the processing and space requirements
enable_satvis: true

# pruning capabilities - must be configured in cron (see documentation)
#   delete_oldest_n - how many oldest captures to delete on each run
#   delete_older_than_n - delete all images older than this many days
delete_oldest_n: 0
delete_older_than_n: 0

# operating system configurations
#   disable_wifi_power_mgmt - if running wireless internet and you want to
#       disable "sleep" mode of your wifi device (assuming it's wlan0), set
#        this to true
disable_wifi_power_mgmt: true

# push processing settings for sending images elsewhere
# Email: Send all images to an email address.
#   NOTE: Make sure you set up your ~/.msmtprc file before enabling email push!
#   enable_email_push - whether to send all images to an external email
#   email_push_address - if enabled, address to send all images to
enable_email_push: false
email_push_address: test@ifttt.com

rto54, 2021-02-15:
[output.log](https://github.com/jekhokie/raspberry-noaa-v2/files/5980320/output.log)

jekhokie, 2021-02-15:
Hi @rto54 - I see in the logs the captures are actually occurring. When you say you don’t have any captures, are you just not seeing images in the webpanel?

Have you checked the /srv/images/ directory to see if the capture images are in that directory?

Also, can you check the database to see what exists in the decoded_passes table?

Finally, have a look at the logs in /var/log/nginx/ to see if there are any suspicious errors for the webpanel.

rto54, 2021-02-15:
Hi @jekhokie,
After launchinf and update again, and a reboot to rpi. I got the images back from the webpanel.

But I think, I have a problemn with my anthenna

Here is a picture :-(

[image]

jekhokie, 2021-02-15:
@rto54 yep that looks like a receiver problem (either `rtl_fm` or the antenna itself). Check your config settings and your antenna to ensure the Bias-Tee is being powered if you have a LNA or similar.

colinluthier, 2021-02-15:
I tried everything to fix this. After about 10hrs of trying to sort out what the problem was I made a fresh SD card. It only took 30min and I was up and running. My old install had a lot on packages that were not installed by the script because they were pre-existing. I think there was a compatibility issue somewhere. I was getting identical images to rto54. I swapped hardware and checked all antenna connections. A fresh install from blank worked. I kept the old SD card and if I get time I will keep looking for the problem.

jekhokie, 2021-02-16:
@rto54 have you tried a fresh install from a pristine raspberry pi os image like Colin did? There are likely library/version conflicts if you built on top of an existing system and/or had pre-installed binaries and/or libraries.

rto54, 2021-02-16:
@jekhokie , Hi, I  installed on a fresh raspian os on a dedicated Rasperry Pi 3b+

jekhokie, 2021-02-16:
@rto54 ok - and you’re still seeing those images? What type of hardware (SDR, LNA, antenna, etc) are you using?

Have you checked that if you have a LNA, it’s being successfully powered during the pass?

rto54, 2021-02-16:
@jekhokie 
_Raspberry PI 3B+ with mSata adaptet and mSata to 240G
_Nooelec NESDR Smart v4 SDR - Prime RTL-SDR or Nooelec R820T2
_Nooelec SAWbird+ GOES Barebones - Premium SAW Filter & Cascaded Ultra-Low Noise LNA Module for NOAA (GOES/LRIT/HRIT/HRPT) Applications. 1688MHz
_Turnstile antenna

I have to go up on the roof to see if evething is goog up there, i'm going to buy a QFH antenna

jekhokie, 2021-02-16:
What I would check is your Bias-Tee setting in `config/settings.yml` (make sure it's set to 'true' if you're wanting to drive using BiasTee) and then check your LNA to ensure it's powered during a pass.

One question - you mentioned a GOES Sawbird filter - is this a notch filter that would actually exclude the 137.xMHz range? If so, you might actually be filtering out the frequency range for capturing NOAA and Meteor-M captures altogether (apologies if I'm misunderstanding what this device is/is doing).

rto54, 2021-02-16:
https://www.nooelec.com/store/sawbird-plus-goes.html

rto54, 2021-02-16:
Sorry is this https://www.nooelec.com/store/sawbird-plus-noaa.html

jekhokie, 2021-02-16:
Ah ok, that makes more sense now (was going to say - GOES isn't captured as part of this framework/requires a completely different setup).

Same advice - I would double check that the bias-tee is actually driving power to your LNA during a capture. I had an issue a while back where for some reason, the LNA wasn't being driven by the Bias-Tee and I had the same output you were getting.

Another thing you could do is try to remove the LNA altogether and see if you get any other FM-based channels with your SDR using `rtl_fm` - that would at least validate the path. Then you could also test receiving without the LNA on the line to see if you get any results.

rto54, 2021-02-16:
I removed the LNA and I put that

```
meteor_receiver: 'rtl_fm'
sdr_device_id: 0
enable_bias_tee: false
receiver_gain: 49.5
receiver_freq_offset: -6

```

I'm waiting for the next passage.

An di will do another test with the same configuration, but with the LNA

jekhokie, 2021-02-16:
Sounds good/looks good - good luck!

colinluthier, 2021-02-16:
@rto54 The only other thing I see is you are maxing out the gain. I get better images at about 2/3rd of max. Type rtl_test from the command line and it will print out allowable gain settings.

rto54, 2021-02-16:
@colinluthier ,Ok it's setting to 15.7 now

jekhokie, 2021-02-19:
@rto54 are you getting images now?

saifdu, 2021-02-20:
Before & After upgrade to v.1.3.0 for NOAA 15

Before:

[image]

After:

[image]

jekhokie, 2021-02-20:
@saifdu do you have any logs you can share (log of the install script output, and log of a capture output)?

colinluthier, 2021-02-20:
This looks like a gain issue. Verify your bias tee is on during a pass if you are using it. Also, check your gain setting in your configuration. To see the gains supported type rtl-test. I had this problem as well. My final solution was to format my SD card and install a fresh copy and set my gain to about two thirds.

saifdu, 2021-02-20:
> 
> 
> @saifdu do you have any logs you can share (log of the install script output, and log of a capture output)?

@jekhokie This my log file, I tried to find the error part, but I don't understand it.  Thanks for your help.

[output.log](https://github.com/jekhokie/raspberry-noaa-v2/files/6015207/output.log)

saifdu, 2021-02-20:
> This looks like a gain issue. Verify your bias tee is on during a pass if you are using it. Also, check your gain setting in your configuration. To see the gains supported type rtl-test. I had this [...]

@colinluthier  I will try it. Thank you for the advice.

saifdu, 2021-02-21:
I have reinstalled and all is well now. Thanks for your support @jekhokie @colinluthier

[image]

jekhokie, 2021-02-21:
Nice, and great capture!

closed 2021-02-21 by jekhokie