jekhokie/raspberry-noaa-v2 #97: Satellite Direction Information

colinluthier, 2021-02-13:
I have missed having the images annotated with the pass direction information. I found it in the log so I am trying to figure out how to catch it but I don't have much experience in bash.

`$WXMAP -T "${SAT_NAME}" -H "${TLE_FILE}" -p 0 ${extra_map_opts} -o "${epoch_adjusted}" $map_overlay >> $NOAA_LOG 2>&1`

This line outputs the wxmap output to the log that is structured like:

```
Satellite: NOAA 18
Pass Start: 2021-02-13 17:27:22 UTC
Pass Duration: 15:14
Elevation: 37
Azimuth: 98
Direction: southbound
```

I am thinking of an approach like:

```
$WXMAP -T "${SAT_NAME}" -H "${TLE_FILE}" -p 0 ${extra_map_opts} -o "${epoch_adjusted}" $map_overlay >>map_out>&1

log "${map_out}"
direction="${map_out#*Direction:}" # take everything after the 'Direction:'
```

But I really don't understand the >> operator at all.

colinluthier, 2021-02-13:
Grabbing the Azimuth would be of interest as well.

colinluthier, 2021-02-13:
azimuth="${map_out#*Azimuth:}" # take everything after the 'Azimuth:'
azimuth="${azimuth#%Direction*} # take everything before 'Direction:'

This might be problematic if we don't strip spaces and CR/NL?

colinluthier, 2021-02-14:
The -f command in predict has the information needed. I am not quite sure how to parse it just yet.

```
The  information  produced  includes  the  date/time in Unix format (the number of seconds
       since midnight UTC on January 1, 1970), the date/time in ASCII (UTC), the elevation of the
       satellite  in  degrees,  the  azimuth  heading of the satellite, the orbital phase (modulo
       256), the latitude and longitude of  the  satellite's  sub-satellite  point  at  the  time
       specified,  the  slant  range  to  the  satellite in kilometers with respect to the ground
       station's  location,  the  orbit  number,  and  the   spacecraft's   sunlight   visibility
       information. 
```
I am putting up the predict calls here so I don't have to keep finding them.

```
# come up with prediction start/end timings for pass
predict_start=$($PREDICT -t $TLE_FILE -p "${OBJ_NAME}" | head -1)
predict_end=$($PREDICT   -t $TLE_FILE -p "${OBJ_NAME}" | tail -1)
max_elev=$($PREDICT      -t $TLE_FILE -p "${OBJ_NAME}" | awk -v max=0 '{if($5>max){max=$5}}END{print max}')
end_epoch_time=$(echo "${predict_end}" | cut -d " " -f 1)
```

jekhokie, 2021-02-16:
Running the predict command with the `-f` switch indeed includes information like elevation and azimuth - here are the columns of the output when it shows the results:

```
Date/Time (Epoch) | Date/Time (ASCII) | Elevation | Azimuth Heading | Orbital Phase | Lat | Lon | Slant Range | Orbit # | Sunlight VIsibility
```
The thing we're missing from the command line I think (unless I just missed it in the man page) is something displayed in the GUI version, which is radial velocity, which combined with azimuth and depending on whether the velocity is positive or negative, can tell us direction of travel (e.g. radial velocity toward base station (+) with azimuth degrees 90>x>270 would be a "northbound" pass/coming from south, etc.).

Theoretically we could get the positional information of the satellite at 2 points close in time (~10 seconds difference) and use the slant range along with azimuth as well - for example, if slant range is positive between points X and X+1, that means the satellite is moving away from the base station, and that info mixed with azimuth could tell us the direction of travel. Again, multiple point calculations, but this might actually be our best bet.

jekhokie, 2021-02-16:
Expanding - the approach here would be like this (thanks Colin for the discussion):

1. Add a column to the `predict_passes` table that includes `azimuth`.
2. Update the schedule script to get the azimuth from the predict output using field 6 of the predict output like so: `predict -t tmp/orbit.tle -p "METEOR-M 2" 1613444323 | head -1 | awk '{ print $6 }'`.
3. Store the azimuth in the database.
4. Add a configurable property to `config/settings.yml` that allows the user to enable/disable showing direction of travel.
5. Update the receive scripts for NOAA and Meteor to use azimuth to determine direction of travel and update annotation of images if desired. Azimuth at start of pass of north of west or north of east means southbound pass, while south of west or south of east means northbound pass.
6. Update the Model in the webpanel to pull the azimuth value.
7. Display the direction of travel as a property in the captures view and also in the predict view next to the elevation.

closed 2021-02-17 by jekhokie