jekhokie/raspberry-noaa-v2 #108: [Error] installing for non "pi" user

mbruski, 2021-02-14:
After running the install_and_upgrade.sh script I get the following error:

Processing triggers for libc-bin (2.28-10+rpi1) ...
 ✓   Ansible install complete!
 * Checking for configuration settings...
 ✓   Settings file ready!
 * Running Ansible to install and/or update your raspberry-noaa-v2...

PLAY [install raspberry-noaa-v2 core] ********************************************************************************************************************************************************************************************************

TASK [get os release] ************************************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [common : noaa environment variables] ***************************************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "checksum": "fbab5a3e66f5ff73e9027e71766d9376ea5724f7", "msg": "Destination directory ~pi does not exist"}
	to retry, use: --limit @/home/not-pi/raspberry-noaa-v2/ansible/site.retry

PLAY RECAP ***********************************************************************************************************************************************************************************************************************************
localhost                  : ok=1    changed=0    unreachable=0    failed=1   

Error:   Something failed with the install - please inspect the logs above

How do I get around this?

jekhokie, 2021-02-14:
@mbruski right now the framework doesn't support other users/assumes that the user account 'pi' is the one being used on the device.

mbruski, 2021-02-14:
IDK but one might assume the env vars $HOME and $USER could be used in the setup script to auto magically configure for the user that is doing the install or is that not possible with ansible?

jekhokie, 2021-02-14:
100 percent - I actually have aspirations to help this topic along eventually because with the exception of a few ARM-specific packages that need to be switched, I don’t actually see why this framework could work on Ubuntu-based (non-pi) systems as well.

dinoj-bg, 2021-02-15:
I was using nico version and now Yours V2 on ubuntu 20.04 on intel CPU platform /amd64 like different user. I have manual download wxtoimg-amd64-2.11.2-beta.deb and compile predict from source and everithing is OK, but everytime when I do git pull I need to edit few files - in ansible manual edit few yml files. I use grep to find which files to edit.   May be if we  set $HOME, $USER and may be $ARCH variables will help. Also is good idea to check php version. I use php7.4 so I need manual edit nginx site file.

colinluthier, 2021-02-16:
From stack-overflow
I think there are several answers given here that would work, but I thought I'd show that you can get this from the ansible user module, by registering it as a variable.

- user:
    name: www-data
    state: present
  register: webserver_user_registered
Note: it will create the user if it doesn't exist...
So we can use debug to show the values of that var, including the path...

- debug:
    var: webserver_user_registered

TASK [wordpress : debug] ******************
ok: [wordpresssite.org] => {
    "webserver_user_registered": {
        "append": false,
        "changed": false,
        "comment": "www-data",
        "failed": false,
        "group": 33,
        "home": "/var/www",      <<------ this is the user home dir
        "move_home": false,
        "name": "www-data",
        "shell": "/usr/sbin/nologin",
        "state": "present",
        "uid": 33
    }
}
And you can use those properties in other modules like this;

- file:
    name: "{{ webserver_user_registered.home }}/.wp-cli"
    state: directory

rustytravis, 2021-03-05:
> After running the install_and_upgrade.sh script I get the following error:
...
"fbab5a3e66f5ff73e9027e71766d9376ea5724f7", "msg": "Destination directory ~pi does not exist"}
...
> How do I get around this?

After editing the config/settings.yml file, the fact that this odroid xu4 Armbian OS has no pi user, no libjpeg9, and no php7.2 invited these changes, which allowed successful completion of the install_and_upgrade.sh script:

`export wai=$(whoami)
sed -i \
  -e "s|/home/pi|/home/$wai|g" \
  -e "s|pi pi|$wai $wai|g" \
  -e "s|~pi|~$wai|g" \
  -e "s|pi:www-data|$wai:www-data|g" \
  -e "s|: pi|: $wai|g" \
  -e "s|libjpeg9|libjpeg62-turbo|g" \
  -e "s|php7.2|php7.3|g" \
$(grep -Erl "/home/pi|pi pi|~pi|pi:www-data|: pi|libjpeg9|php7.2" ./*)
`

mbruski, 2021-03-06:
@rustytravis thank you kindly for the script.  It made all the necessary changes to replace the default user and I was able to do a partial install.  I just wasn't prepared for all the changes that the authors have committed over the last 2 weeks.  This thing is getting way too complicated for a regular user to manage/configure so I'm going to throw in the towel and move on.  For now, I get way better lrpt/apt images produced on my Windows setup with SDR# even if it does require a small amount of human intervention.  Oh well... at least I tried.

dom-robinson, 2021-09-29:
Looks like some demand and requirement for this and it makes sense. In absence of @jekhokie im flagging this 'help wanted' to see if the community can generate a workign solution.

rustytravis, 2021-09-30:
On 9/29/21 2:23 AM, Dom Robinson wrote:
>
> Looks like some demand and requirement for this and it makes sense. In 
> absence of @jekhokie <https://github.com/jekhokie> im flagging this 
> 'help wanted' to see if the community can generate a workign solution.
>

cd into the user's $HOME/raspberry-noaa-v2 directory. After editing the 
config/settings.yml file, the fact that this armhf sbc has no "pi" user 
invited this one liner, which allowed successful completion of the 
./install_and_upgrade.sh script:

wai=$(whoami)
sed -i \
-e "s|/home/pi|/home/$wai|g" \
-e "s|pi pi|$wai $wai|g" \
-e "s|~pi|~$wai|g" \
-e "s|pi:www-data|$wai:www-data|g" \
-e "s|: pi|: $wai|g" \
-e "s|(pi)|($wai)|g" \
-e "s| pi user | $wai user |g" \
-e "s| pi | $wai |g" \
-e "s|\"sudo |\"sudo -n |" \
$(grep -Erl "/home/pi|pi pi|~pi|pi:www-data|: pi|libjpeg9|php7.2|(pi)| 
pi user | pi |\"sudo " ./*)

dom-robinson, 2021-09-30:
Very kind - thats awesome!

So for my clarity this 'one liner' is pasted at the commandline? 

I am wondering if this should be included as an annexe to the main readme documentation under the troubleshooting or something similar?

Cadair, 2021-10-05:
There is no reason why the whole repo can't be made to be user independent. I don't think it should be that complex, there's environment variables and ansible trickery for getting the current user. (i.e. we install as the user who runs the script).

I am not sure that recommending someone run sed on their checkout of the repo is a good idea in our docs as it will cause them major issues with updating.

mihajlo2003petkovic, 2023-10-04:
This was fixed and currently works for any Pi user and also works on Debian Bullseye based distros. I've tested it on LMDE 5 which us based on Bullseye (version 6 is based on Bookworm and doesn't work). I'll close this issue for now and will reopen it in the future if there is need to.

closed 2023-10-04 by mihajlo2003petkovic

rustytravis, 2023-10-04:
On 10/4/23 01:55, MihajloPi wrote:
>
> This was fixed and currently works for any Pi user and also works on 
> Debian Bullseye based distros. I've tested it on LMDE 5 which us based 
> on Bullseye (version 6 is based on Bookworm and doesn't work). I'll 
> close this issue for now and will reopen it in the future if there is 
> need to.
>

Thank you. Many of us are using Debian 12 Bookworm on our Rpi(s).