jekhokie/raspberry-noaa-v2 #112: Add a Config Checker Script to Installer

jekhokie, 2021-02-15:
It can be difficult to compare new settings from `config/settings.yml.sample` on upgrades, especially when there are a lot of new configs. Have a pre-script check (Python script that can parse settings) put into the front of the `install_and_upgrade.sh` script to check that the `config/settings.yml` has all the settings required and, if not, produce an error with the missing settings, informing the user of what needs to be put into the file as part of the upgrade.

colinluthier, 2021-02-15:
Or if it put in the missing items at default settings?

jekhokie, 2021-02-15:
@colinluthier I'd considered just "doing the alignment" but I'm really hesitant about switching things out from under the users, especially when it comes to their own config settings. If there were a bug in the script, or if in the future there are more sensitive items that require the config.yml to be locked down, I'd rather not permit the framework to have access to it in general.

colinluthier, 2021-02-15:
That's a good point, especially if we are adding FTP or email passwords to settings.yml.

closed 2021-02-16 by jekhokie