jekhokie/raspberry-noaa-v2 #115: [Error]Sat elevation not getting generated correctly

colinluthier, 2021-02-15:
I am not getting any output of the sat_max_elevation to the annotation. I did a quick check for typos. I will check again tonight.

colinluthier, 2021-02-15:
annotation="${SAT_NAME} $enhancement ${capture_start} Elev: $SAT_MAX_ELEVATION°"

should be:

annotation="${SAT_NAME} $enhancement ${capture_start} Elev: ${SAT_MAX_ELEVATION}°"

colinluthier, 2021-02-15:
The same change needs to be made to recieve_meteor.sh.

jekhokie, 2021-02-15:
Nice catch! I’ll make the update this evening.

There must be some kind of nuance to this because my images actually show the elevation fine (as are Dom’s images). But it doesn’t hurt to use the alternative notation if it helps.

colinluthier, 2021-02-15:
I don’t fully understand the notation at all. My programming experience is in a dead language called macro11 and it was much more pedantic.

jekhokie, 2021-02-16:
This has likely been fixed as part of the PR merged associated with Issue #118 - awaiting feedback.

colinluthier, 2021-02-16:
This is confirmed to work now. Thank you!

closed 2021-02-16 by colinluthier