jekhokie/raspberry-noaa-v2 #125: [Error]  Gmail

klikis, 2021-02-16:
Hi everyone, although I have configured the gmail server to send an email for receipts I can't get it to work. Where is my error?
[image]
[image]
feb 16 22:05:31 host=smtp.gmail.com tls=on auth=on user=gabrielesomma from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com errormsg='the server sent an empty reply' exitcode=EX_PROTOCOL
feb 16 22:08:49 host=smtp.gmail.com tls=on auth=on user=gabrielesomma from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com smtpstatus=535 smtpmsg='535-5.7.8 Username and Password not accepted. Learn more at\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials y29sm3209301edl.91 - gsmtp' errormsg='authentication failed (method PLAIN)' exitcode=EX_NOPERM
feb 16 22:10:25 host=smtp.gmail.com tls=on auth=on user=gabriele somma from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com smtpstatus=535 smtpmsg='535-5.7.8 Username and Password not accepted. Learn more at\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials a11sm4878482ejt.112 - gsmtp' errormsg='authentication failed (method PLAIN)' exitcode=EX_NOPERM
feb 16 22:15:04 host=smtp.gmail.com tls=on auth=on user=gabriele somma from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com smtpstatus=535 smtpmsg='535-5.7.8 Username and Password not accepted. Learn more at\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials r1sm14542515eds.70 - gsmtp' errormsg='authentication failed (method PLAIN)' exitcode=EX_NOPERM
feb 16 22:17:44 host=smtp.gmail.com tls=on auth=on user=gabrielesomma@gmail.com from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com errormsg='the server sent an empty reply' exitcode=EX_PROTOCOL
feb 16 22:18:29 host=smtp.gmail.com tls=on auth=on user=gabrielesomma@gmail.com from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com smtpstatus=535 smtpmsg='535-5.7.8 Username and Password not accepted. Learn more at\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials g23sm14749188ejd.27 - gsmtp' errormsg='authentication failed (method PLAIN)' exitcode=EX_NOPERM
feb 16 22:19:55 host=smtp.gmail.com tls=on auth=on user=gabriele somma from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com errormsg='the server sent an empty reply' exitcode=EX_PROTOCOL
feb 16 22:25:33 host=smtp.gmail.com tls=off auth=on user=gabriele somma from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com errormsg='the server sent an empty reply' exitcode=EX_PROTOCOL
feb 16 22:26:30 host=smtp.gmail.com tls=off auth=on user=gabrielesomma@gmail.com from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com errormsg='the server sent an empty reply' exitcode=EX_PROTOCOL
feb 16 22:27:31 host=smtp.gmail.com tls=off auth=off from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com errormsg='the server sent an empty reply' exitcode=EX_PROTOCOL
feb 16 22:28:08 host=smtp.gmail.com tls=on auth=off from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com errormsg='the server sent an empty reply' exitcode=EX_PROTOCOL
feb 16 22:28:43 host=smtp.gmail.com tls=on auth=off from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com smtpstatus=530 smtpmsg='530-5.7.0 Authentication Required. Learn more at\n530 5.7.0  https://support.google.com/mail/?p=WantAuthError f11sm5950461eje.107 - gsmtp' errormsg='envelope from address gabrielesomma@gmail.com not accepted by the server' exitcode=EX_DATAERR
feb 16 22:29:28 host=smtp.gmail.com tls=on auth=on user=gabrielesomma@gmail.com from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com smtpstatus=535 smtpmsg='535-5.7.8 Username and Password not accepted. Learn more at\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials cw9sm5225434edb.2 - gsmtp' errormsg='authentication failed (method PLAIN)' exitcode=EX_NOPERM
feb 16 22:35:03 host=smtp.gmail.com tls=on auth=on user=gabrielesomma@gmail.com from=gabrielesomma@gmail.com recipients=gabrielesomma@gmail.com smtpstatus=535 smtpmsg='535-5.7.8 Username and Password not accepted. Learn more at\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials y1sm14615505edq.26 - gsmtp' errormsg='authentication failed (method PLAIN)' exitcode=EX_NOPERM

klikis, 2021-02-16:
[image]
[image]

jekhokie, 2021-02-17:
As discussed, send me your latest msmtprc file that you tested using port 587 and user specified as `gabrielesomma`.

jekhokie, 2021-02-18:
@klikis are you still having issues with this?

closed 2021-02-19 by jekhokie