jekhokie/raspberry-noaa-v2 #131: Discord Auto-Post of Captures

jekhokie, 2021-02-17:
Add ability to post image captures to Discord channel

colinluthier, 2021-02-17:
IFTT also will do discord if we don't want to figure out the radioguy script.

jekhokie, 2021-02-17:
I actually disregarded the script he sent - it's MUCH simpler using a bash script and just a curl command. I have a branch in place and working already - leaving it overnight to make sure the morning passes do what I expect, and if they do, going to merge tomorrow.

jekhokie, 2021-02-17:
This functionality is integrated in a branch and will be running for the next set of captures I have scheduled - if everything goes well and the images show up for both meteor and noaa captures, I will merge the branch (likely tomorrow AM).

colinluthier, 2021-02-17:
He didn’t send the script, I stole it from a discord group we share. ;)

closed 2021-02-17 by jekhokie