jekhokie/raspberry-noaa-v2 #132: [Error]

colinluthier, 2021-02-17:
**Describe the bug**
A git pull failed to update the database, or that's what it looks like.
**Screenshots and/or logs**
When calling up the pass list from the web:

```
Warning: SQLite3::query(): Unable to prepare statement: 1, no such column: pass_start_azimuth in /var/www/wx-new/App/Models/Pass.php on line 19

Fatal error: Uncaught Error: Call to a member function fetchArray() on boolean in /var/www/wx-new/App/Models/Pass.php:24 Stack trace: #0 /var/www/wx-new/App/Controllers/PassesController.php(10): App\Models\Pass->getList() #1 [internal function]: App\Controllers\PassesController->indexAction(Array) #2 /var/www/wx-new/Lib/Controller.php(22): call_user_func_array(Array, Array) #3 /var/www/wx-new/Lib/Router.php(56): Lib\Controller->__call('index', Array) #4 /var/www/wx-new/Lib/Router.php(13): Lib\Router->handleRequest() #5 /var/www/wx-new/public/index.php(16): Lib\Router->__construct() #6 {main} thrown in /var/www/wx-new/App/Models/Pass.php on line 24
```

jekhokie, 2021-02-17:
@colinluthier the git pull won't update the database. The sequence for upgrading is this:

```
git pull
# edit config/settings.yml to include new settings from config/settings.yml.sample
./install_and_upgrade.sh
```

Try that sequence - it should bring you back in line as it will run the database migrations and add the missing columns as well.

colinluthier, 2021-02-17:
```
pi@raspberrypi:~/raspberry-noaa-v2 $ git pull
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Total 5 (delta 4), reused 5 (delta 4), pack-reused 0
Unpacking objects: 100% (5/5), done.
From https://github.com/jekhokie/raspberry-noaa-v2
   ef9f0c2..ab712b7  master     -> origin/master
Updating ef9f0c2..ab712b7
Fast-forward
 scripts/receive_meteor.sh | 2 +-
 scripts/receive_noaa.sh   | 2 +-
 2 files changed, 2 insertions(+), 2 deletions(-)
You have new mail in /var/mail/pi
pi@raspberrypi:~/raspberry-noaa-v2 $ bash install_and_upgrade.sh
 * Checking configuration files...
Configurations 'config/settings.yml.sample' and 'config/settings.yml' aligned
 ✓   Config check complete!
 * Installing Python dependencies...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: envbash==1.2.0 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/raspberry-noaa-v2/requirements.txt (line 1)) (1.2.0)
Requirement already satisfied: ephem==3.7.7.1 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/raspberry-noaa-v2/requirements.txt (line 2)) (3.7.7.1)
Requirement already satisfied: idna==2.8 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/raspberry-noaa-v2/requirements.txt (line 3)) (2.8)
Requirement already satisfied: numpy==1.18.1 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/raspberry-noaa-v2/requirements.txt (line 4)) (1.18.1)
Requirement already satisfied: Pillow==7.1.2 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/raspberry-noaa-v2/requirements.txt (line 5)) (7.1.2)
Requirement already satisfied: pyyaml==5.4.1 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/raspberry-noaa-v2/requirements.txt (line 6)) (5.4.1)
Requirement already satisfied: requests==2.25.1 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/raspberry-noaa-v2/requirements.txt (line 7)) (2.25.1)
Requirement already satisfied: urllib3==1.26.3 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/raspberry-noaa-v2/requirements.txt (line 8)) (1.26.3)
Requirement already satisfied: chardet<5,>=3.0.2 in /usr/lib/python3/dist-packages (from requests==2.25.1->-r /home/pi/raspberry-noaa-v2/requirements.txt (line 7)) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests==2.25.1->-r /home/pi/raspberry-noaa-v2/requirements.txt (line 7)) (2018.8.24)
 ✓   Successfully aligned required Python packages!
 * Checking for configuration settings...
 ✓   Settings file ready!
 * Running Ansible to install and/or update your raspberry-noaa-v2...

PLAY [install raspberry-noaa-v2 core] ******************************************************************************************

TASK [get os release] **********************************************************************************************************
ok: [localhost]

TASK [common : noaa environment variables] *************************************************************************************
ok: [localhost]

TASK [common : predict directory] **********************************************************************************************
ok: [localhost]

TASK [common : predict config file] ********************************************************************************************
ok: [localhost]

TASK [common : wxtoimg config file] ********************************************************************************************
ok: [localhost]

TASK [common : wxtoimg license terms acceptance file] **************************************************************************
ok: [localhost]

TASK [common : log directory for raspberry-noaa-v2 logs] ***********************************************************************
ok: [localhost]

TASK [common : logrotate settings for raspberry-noaa-v2 logs] ******************************************************************
ok: [localhost]

TASK [common : gnuradio home directory] ****************************************************************************************
ok: [localhost]

TASK [common : gnuradio prefs directory] ***************************************************************************************
ok: [localhost]

TASK [common : gnuradio prefs file] ********************************************************************************************
ok: [localhost]

[... 190 lines cut ...]
Installing dependencies (including require-dev) from lock file
Package operations: 3 installs, 0 updates, 0 removals
  - Installing symfony/polyfill-mbstring (v1.22.0): Loading from cache
  - Installing symfony/polyfill-ctype (v1.22.0): Loading from cache
  - Installing twig/twig (v3.2.1): Loading from cache
Generating autoload files
 * Scheduling first passes for imagery...
INFO : Clearing and re-creating TLE file with latest...
INFO : Clearing existing scheduled 'at' capture jobs...
INFO : Scheduling new capture jobs...
INFO : Done scheduling jobs!
 * First passes scheduled!

-------------------------------------------------------------------------------
 CONGRATULATIONS!

 raspberry-noaa-v2 has been successfully installed/upgraded!
 You can view the webpanel updates by visiting the following URL in a web browser:
 http://<YOUR_IP_OR_HOSTNAME>:<CONFIGURED_PORT>/

 You can also see the URL listed in the 'output web server url' play output above.
-------------------------------------------------------------------------------

pi@raspberrypi:~/raspberry-noaa-v2 $ sudo apt-get install sqlitebrowser
```

colinluthier, 2021-02-17:
This error?

```
INFO :   - applying 03_add_pass_start_azimuth.sql
/home/pi/raspberry-noaa-v2/db_migrations/update_database.sh: line 50: /home/pi/raspberry-noaa-v2/db_migrations/03_add_pass_start_azimuth.sql: No such file or directory
```

jekhokie, 2021-02-17:
The issue is in the `update_database.sh` script - it is referring to `03_add_pass_start_azimuth.sql` but the actual migration script name is `03_add_pass_azimuth_direction.sql`. I just fixed this and pushed an update to master - please `git pull` and then re-run the install script.

jekhokie, 2021-02-17:
Confirmed fix worked as expected - this is no longer an issue.

closed 2021-02-17 by jekhokie