jekhokie/raspberry-noaa-v2 #135: install script error

tomamdy, 2021-02-17:
I get this error.

Traceback (most recent call last):
  File "scripts/tools/diff_yaml_configs.py", line 13, in <module>
    import yaml
ModuleNotFoundError: No module named 'yaml'
Error:   Please update your config/settings.yml file to include the missing parameters from config/settings.yml.sample

jekhokie, 2021-02-17:
@tomamdy nice catch - just pushed a change. Please "git pull" then re-run the installer and let me know how that goes.

jekhokie, 2021-02-17:
Just caught one more thing and pushed another update that should address the install.

jekhokie, 2021-02-18:
I did a complete reinstall from scratch using the changes in this submission and it worked correctly - I'm going to close this, but feel free to re-open if you do in fact still have a blocker/run into issues. Thanks!

closed 2021-02-18 by jekhokie