jekhokie/raspberry-noaa-v2 #144: Cannot remove  meteor *.bmp

tomamdy, 2021-02-18:
From pi@raspberrypi.fritz.box Wed Feb 17 20:34:48 2021
Return-path: <pi@raspberrypi.fritz.box>
Envelope-to: pi@raspberrypi.fritz.box
Delivery-date: Wed, 17 Feb 2021 20:34:48 +0100
Received: from pi by jeknoaatest with local (Exim 4.92)
	(envelope-from <pi@raspberrypi.fritz.box>)
	id 1lCSb2-0000ZB-Hd
	for pi@raspberrypi.fritz.box; Wed, 17 Feb 2021 20:34:48 +0100
Subject: Output from your job        6
To: pi@raspberrypi.fritz.box
Message-Id: <E1lCSb2-0000ZB-Hd@jeknoaatest>
From: pi@raspberrypi.fritz.box
Date: Wed, 17 Feb 2021 20:34:48 +0100

INFO : The system doesn't have enough space to store a Meteor pass on RAM
INFO : I'll flip this image pass because FLIP_METEOR_IMG is set to true
INFO : Starting rtl_fm record
INFO : Demodulation in progress (QPSK)
INFO : Producing spectrogram
INFO : Decoding in progress (QPSK to BMP)
INFO : I got a successful METEOR-M-2-20210217-194556.dec file. Decoding APID 68
INFO : Rectifying image to adjust aspect ratio
rm: cannot remove '/srv/images/METEOR-M-2-20210217-194556-122.bmp': No such file or directory
rm: cannot remove '/srv/audio/meteor/METEOR-M-2-20210217-194556.bmp': No such file or directory

jekhokie, 2021-02-18:
It almost looks like there was no image processed (rectified) for that pass? What does your `/var/log/raspberry-noaa-v2/output.log` show you? Did you actually get an image output?

tomamdy, 2021-02-18:
Yes Justin, there was no image processed. That means that it was receiving related and no error ?

jekhokie, 2021-02-18:
That error likely means the audio file (received audio) didn't have enough data to produce an image.

closed 2021-02-18 by tomamdy