jekhokie/raspberry-noaa-v2 #148: Add French Translation to Webpanel

jekhokie, 2021-02-18:
Add the ability to have a French translation for the webpanel.

rto54, 2021-02-18:
[fr_V1.0_100221.txt](https://github.com/jekhokie/raspberry-noaa-v2/files/6004683/fr_V1.0_100221.txt)

rto54, 2021-02-18:
[fr_V1.0_180221.txt](https://github.com/jekhokie/raspberry-noaa-v2/files/6004694/fr_V1.0_180221.txt)

closed 2021-02-18 by jekhokie

rto54, 2021-02-23:
@jekhokie , Good Evening, there was an error on line 11, here is the new version 
[fr_V1.10_230221.txt](https://github.com/jekhokie/raspberry-noaa-v2/files/6032030/fr_V1.10_230221.txt)

jekhokie, 2021-02-24:
@rto54 this is all set with the latest merge to master branch - do a `git pull` and then re-run the `./install_and_configure.sh` script and you should be all set.

rto54, 2021-02-24:
@jekhoki, it's Ok, thanks