jekhokie/raspberry-noaa-v2 #150: [Error]

ltiisidii, 2021-02-18:
**Describe the bug**
Try to install from scratch and an existing one, in both it gives the error when installing the following packages:

libjpeg9
libjpeg9-dev
predict

**Screenshots and/or logs**

```
TASK [common : enable systemd wifi power management disable service] ***************************************************************************************************
skipping: [localhost]

TASK [common : install dependencies] ***********************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'libjpeg9' is available"}
        to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry

PLAY RECAP *************************************************************************************************************************************************************
localhost                  : ok=11   changed=10   unreachable=0    failed=1

Error:   Something failed with the install - please inspect the logs above
```

**Server information (please complete the following information):**
 - OS: Debian GNU/Linux 10 (Debian Buster with Raspberry Pi Desktop)
 - Python version 3.7.3
 - SDR device RTL-SDR v3

Additional:

cat /etc/apt/sources.list
deb http://ftp.debian.org/debian/ buster main contrib non-free
deb http://security.debian.org/ buster/updates main contrib non-free
deb http://ftp.debian.org/debian/ buster-updates main contrib non-free

[Packages-debian: Predict](https://packages.debian.org/search?suite=buster&keywords=predict)

[Packages-debian: libjpeg9](https://packages.debian.org/search?suite=buster&section=all&arch=any&searchon=names&keywords=libjpeg9)

[Packages-debian: libjpeg9-dev](https://packages.debian.org/search?suite=buster&section=all&arch=any&searchon=names&keywords=libjpeg9-dev)

jekhokie, 2021-02-19:
@ltiisidii I see that Buster uses libjpeg62-turbo - this likely will require some changes to support buster.

ltiisidii, 2021-02-20:
Yes, I fixed it by installing libjpeg62-turbo and libjpeg62-turbo-dev.

Predict installation:

```
https://www.qsl.net/kd2bd/predict.html
wget https://www.qsl.net/kd2bd/predict-2.2.7.tar.gz
./configure
make
make install 
```

So far it works "well" (I live in an apartment and the balcony overlooks the entire East, overlooking the north and south), with:
RTL-SDR v3 (bias tee activated)
RTL-SDR Blog Wideband Low Noise Amplifier Bias Tee Powered
Flamingo FM (FM Notch Filter)
V dipole (kit rtl-sdr v3) at 120 ° pointing south 

Noaa 18 (Gain: 29.7)
[image]

Noaa 15 (Gain: 15,7)
[image]

Meteor M2:
[image]

jekhokie, 2021-02-21:
@ltiisidii interesting - were those the only 2 packages you needed to install (replacing the libjpeg packages) and everything else "just worked" without changes? If so, I can update the code to support Buster by installing the turbo packages for Buster, and non-turbo packages for other distros.

ltiisidii, 2021-02-23:
@jekhokie - Let me try this week with a vm, with a clean image because I had the v1 version installed on the same that I installed the v2.
And the problem with the lack of the predict package I had with debian buster.

ltiisidii, 2021-02-25:
Installation on a scratch image by removing
      - libjpeg9
      - libjpeg9-dev
      - predict
      
in the File:
ansible/roles/common/tasks/dependencies.yml

I finish ok and the page lifts ok.

To install predict I had to install:
sudo apt-get install libasound2-dev
wget https://www.qsl.net/kd2bd/predict-2.2.7.tar.gz 
tar xvzf predict-2.2.7.tar.gz
cd predict-2.2.7
./build

So far I can not make it load the past of the satellites

jekhokie, 2021-02-28:
@ltiisidii just checking in - did the install work for you the way you expected it to/are you getting captures? Can you post the exact sequence of steps you followed for the install?

jekhokie, 2021-03-04:
Closing for now - if this is still an issue that makes sense to include/has a solid path forward, we can re-open and consider.

closed 2021-03-04 by jekhokie