jekhokie/raspberry-noaa-v2 #156: Timezone in webpage

tomamdy, 2021-02-19:
Can get the correct time for Amsterdam in the webpanel, but it still gives America/New York

[image]

# locale settings for timezone and language
#   timezone: see https://www.php.net/manual/en/timezones.php
#   lang_setting: see the 'webpanel/App/Lang' folder for available
#                 languages (2-letter filename - e.g. ar, bg, de, en, es, nl, sr)
timezone:Europe/Amsterdam
lang_setting: en

jekhokie, 2021-02-19:
@tomamdy did you re-run the install_and_upgrade.sh script after making the config change? What does the file /var/www/wx-new/Config/Config.php have for the locale? If it’s still America/New York, it likely means you updated the settings.yml file but haven’t yet re-run the install script since making the change to have the config pushed to all locations.

tomamdy, 2021-02-20:
It is very strange. With the Raspberry 4 it gives not a problem, but it only happened with the 3B.
Now I changed by hand /var/www/wx-new/Config/Config.php the locale and it is ok now.

closed 2021-02-20 by tomamdy