jekhokie/raspberry-noaa-v2 #157: Feature request about the coordinates in Satvis

dom1706, 2021-02-19:
Hello again, 
I've a little problem with satvis.
On the passes page, if if i click randomly on a satellite arond the Earth, I obtain the values correctly.
But if I click on the satellite name (1st column), and after that if I click on the little white dot above the satellite name, I have an error message "No ground station set".
Where can I put my lat/long for satvis ??
Thanks in advance for your help.

jekhokie, 2021-02-19:
Note to self - this is a task to take the user-defined lat and lon and have Ansible drop them in the web config file, and then have the satvis url templated to include the coordinates as the url encoded get parameters.

jekhokie, 2021-02-20:
This is actually a bit different of an issue - the problem is when the link is clicked, it brings you to the URL with the satellite in the GET parameters but those parameters are missing the lat/lon for the redirected link. Pretty easy fix.

closed 2021-02-20 by jekhokie