jekhokie/raspberry-noaa-v2 #160: Option to use MeteorDemo to process the MeteorM2 and add maps

dom-robinson, 2021-02-20:
https://github.com/Digitelektro/MeteorDemod provides and excellent way to process M2 adding maps.

It requires opencv. The attached opencvinstall3.sh will install both opencv and MeteorDemod.

Once installed you can call meteor demod like this:

meteordemod -t weather.tle -i METEOR-M220210122-073447.s -f jpg -d 22-01-2021

. 

It uses a '.s' file as a source - so ostensibly starts the same as the current gnuradio script, but the processing is different (better?)

At install MeteorDemod uses up a settings.ini in the /resources folder. Settings from RN2 will need to be configured in this to overlay the correct text on the output images.

I attach the example settings.ini too.
[install and settings.zip](https://github.com/jekhokie/raspberry-noaa-v2/files/6015539/install.and.settings.zip)

colinluthier, 2021-02-20:
I deleted my opencv and tried yours but the build failed. I replaced the cmake line with:

```
cmake -D CMAKE_BUILD_TYPE=RELEASE \
        -D CMAKE_INSTALL_PREFIX=/home/pi/installation/OpenCV-"$cvVersion" \
        -D INSTALL_C_EXAMPLES=ON \
        -D INSTALL_PYTHON_EXAMPLES=ON \
        -D WITH_TBB=ON \
        -D WITH_V4L=ON \
        -D OPENCV_PYTHON3_INSTALL_PATH=$cwd/OpenCV-$cvVersion-py3/lib/python3.5/site-packages \
        -D WITH_QT=ON \
        -D WITH_OPENGL=ON \
        -D OPENCV_EXTRA_MODULES_PATH=../../opencv_contrib/modules \
        -D BUILD_EXAMPLES=ON ..
```
Now it is building.

colinluthier, 2021-02-20:
I finally got this all to build. My final make installed these files:

```
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/meteordemod
-- Set runtime path of "/usr/local/bin/meteordemod" to ""
-- Installing: /home/pi/.config/meteordemod
-- Installing: /home/pi/.config/meteordemod/settings.ini
-- Installing: /home/pi/.config/meteordemod/thermal_ref.bmp
-- Installing: /home/pi/.config/meteordemod/ShapeFiles
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_10m_geography_regions_elevation_points.shp
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_110m_graticules_10.shp
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_populated_places.shp
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_110m_graticules_10.shx
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_rivers_lake_centerlines.dbf
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_110m_graticules_10.dbf
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_populated_places.dbf
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_10m_geography_regions_elevation_points.shx
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_populated_places.shx
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_10m_geography_regions_elevation_points.dbf
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_lakes.dbf
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_coastline.shp
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_admin_0_boundary_lines_land.shx
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_admin_0_boundary_lines_land.shp
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_coastline.dbf
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_admin_0_boundary_lines_land.dbf
-- Installing: /home/pi/.config/meteordemod/ShapeFiles/ne_50m_coastline.shx
-- Installing: /home/pi/.config/meteordemod/thermal_ref2.bmp
-- Installing: /home/pi/.config/meteordemod/rain.bmp
```

I think we can do that without compiling from source. Only the settings.ini file needs configuration. As a minimum it needs lat and long.

dom-robinson, 2021-02-21:
hey guys - just got my install.sh tidied up and tested on a clean build on pi4. Version attached. 

[opencvinstall4.sh.zip](https://github.com/jekhokie/raspberry-noaa-v2/files/6017990/opencvinstall4.sh.zip)

tomamdy, 2021-02-24:
is it possible to implement this in raspberry-noaa-v2?

jekhokie, 2021-02-24:
@tomamdy we definitely will. Dom has done most of the work to make this "functional", but time is the enemy around here :-)

dom-robinson, 2021-02-25:
OK I think at this point it needs more hive mind, but the attached should at least get the images on your disc, That said there are some key shortcomings which i will describe as i write:

- SHORTCOMINGS -
-  At the moment the old receive_meteor.sh is set aside as receive_meteor.met_det.sh (so i can switch between the working demodulation and the meteordemod one) 
- I then ```mv receive_meteordemod.sh```  into play as receive_meteorl.sh to test. This means that in config we need to merge both and have a setting switch to select. They can both be merged into receive_meteor together forking just after the .s file is created in the first step of the 'gnuradio' method in the current script. (line 171 in receive_meteor.meteordemod attached)
- Image processing is treated differently. All the enhancements are produced in a single step. However the default -mercator.jpg is massive (50Mb+ and overloads the processing) so i delete that upfront.
- After that files are (supposedly) copied to the right directories : I am fighting the location when RAMFS is full since it defaults to the audio folder for processing and that subsequently makes it less simple to copy all the output files apart from the audio .s file and others away. However i have worked in a manual mv list that does that. I am sure Justin will refactor that much better than i ever could!
- I then mogrify ('batch convert' the copies of the images in the thumb folder to make them smaller thumnail size for longer term use.
- The  DB updates ok. 
- BUT /var/www/wx-new/App/Models/Capture.php lists wrong ehnhancements - so i did manually create an extra line in Captures.php to show the meteordemod outputs but its currently either or - otherwise you get empty boxes on the main webpanel - Given i can see i am working in a change that should be done at a more fundamental level to work with config i figure i should stop work on that bit. I will include my Captures.php with meteordemod enhancement list commented out (line 66)- have a look - its fairly obvious which lines need comment / uncomment to trick it into display at the moment. These should be config.
[Meteor Demod versioing.zip](https://github.com/jekhokie/raspberry-noaa-v2/files/6044031/Meteor.Demod.versioing.zip)

Finally there is a bigger issue with meteordemod. The /home/pi/./config/meteordemod/settings.ini is where any RN2 config related to meteordemo should be written. However whatever i try to put in the groundstation (which is the only lat / long setting in the setup) i am getting maps from about 45 degrees latitude lower than my location. I actually think the ground station lat / long is exactly what is should be, and has no bearing on the map overlay - in which case something in the time formatting or the actual .s file data is incompatible at the moment.

I figure lets get a few more looking and testing it and see if i am getting something wrong.

Files attached. Capture.php, settings.ini

First - I have an RPI4 image i always start from, with GNURadio, GQRX and OpenWebRX already installed and launched on command.

I then pulled latested raspberry noaa two image from justin's repo.

After that i ran my opencv install v4 script (as shared [above](https://github.com/jekhokie/raspberry-noaa-v2/files/6017990/opencvinstall4.sh.zip))

That seems to work, but milage may vary. The working test is to type 'meteordemod' at the commandline and see if you get :

colinluthier, 2021-02-26:
I finally got to read through this. I like it. I think the mv commands are a good way to adapt to the database. Two minor things:

In receivemeteor the lines:

`rm -f ${RAMFS_AUDIO_BASE}.dec` 
This should have a switch. Its a way better way to store the pass than keeping the audio. 

 ``` else
    log "Did not get a successful .bmp image - stopping processing" "ERROR" 
```

should be 

``` else
        log "Did not get a successful Meteor image - stopping processing" "ERROR"```
since it is a jpg and not a bmp.

jekhokie, 2021-03-10:
Note to self re: getting meteordemod installed - it would be best to distribute a pre-built meteordemod binary with this framework vs. building OpenCV (required) from scratch to build the meteordemod package, which can take upwards of 2 hours in worst case (best case, still ~20 minutes). I've opened an issue [here]() inquiring if the author is open to this.

If they are, then it's as simple as the following (likely Ansible install is best):
1. ` sudo apt-get -y install libopencv-shape3.2 libopencv-stitching3.2 libopencv-superres3.2 libopencv-videostab3.2 libopencv-contrib3.2`
2. Place pre-built meteordemod binary somewhere.
3. Create a `~pi/.config/meteordemod` directory.
4. Place a customized `~pi/.config/meteordemod/settings.ini` file into place.
5. Git clone the meteordemod repository.
6. Copy everything from `MeteorDemod/resources/*` to `~pi/.config/meteordemod/`.
7. Update scripts to use the command sequence, which looks like so: `./meteordemod -i METEOR-M220210122-073447.s -t /home/pi/raspberry-noaa-v2/tmp/orbit.tle -o /tmp/ -f jpg --date 01-22-2021`
8. Update web interface and database appropriately.

jekhokie, 2021-03-12:
@dom-robinson did you ever get to the point of having map overlays work correctly with this? I’d be happy to start on this but I just don’t want to put a ton of effort in if we’re still needing to sort out the map overlay issue with the developer. Thanks!

colinluthier, 2021-03-15:
My understanding is the map works with a fresh lte.

dom-robinson, 2021-09-29:
CLosing here and redirecting to the in progress PR #439 since we are 'nearly there' - and this job is in the queue as the next major to-do after the current gnuradio effort.

closed 2021-09-29 by dom-robinson