jekhokie/raspberry-noaa-v2 #181: Raspberry Noaa V2 no decode

maxisantilli, 2021-02-27:
Hello Justin how are you?

I'm Maximiliano, I tell you that I installed raspberry noaa v2 to the letter but I can't make it work.
It does not decode anything.
The web works well, the rtl too, I don't know what else to try.
  : /

I'm new to linux
you can help me?
Greetings!!

jekhokie, 2021-02-28:
Hi @maxisantilli - can you join us on Discord chat so we can help you? There are many of us in the discord chat that are helping each other, it’s likely your best next step. Here is an invite to join us: https://discord.gg/FFGDujau

maxisantilli, 2021-02-28:
I solved the problem !!

If you put the gain in automatic, do not put: 0

is: 0.0

That's why I was running the sdr!

[image]

here the result.

Helix antenna

without lna
no fm filter

only one antenna tuner

jekhokie, 2021-02-28:
Nice find - this is likely a bug that needs to be addressed. My suspicion is that the gain parameter needs to be included even if the user specifies auto-gain (0), and including 0.0 results in the conditional passing the zero value (somewhat of a red herring). It’s likely best to always pass the gain parameter, even if it’s zero - this will provide better consistency in the parameter passing. Marking this as a bug to be fixed.

closed 2021-02-28 by jekhokie