jekhokie/raspberry-noaa-v2 #183: Pi 3B+, Buster, No decoded images after first decode.

royldean, 2021-02-28:
I have two Pi's set up with RPi-Noaa-V2, each with it's own RTL-SDR and Antenna, running identical installs (both git pulled on Friday night).

One of them works perfectly normally - schedules observations and resultant decoded images are listed as expected.    The second, however, only decoded ONE image when it was first installed, and now doesn't show any decodes - and subsequent git pulls and installs seem to do nothing to improve the situation.   Note that it DOES update the scheduled observations perfectly, including "greying out" the past observations.... It just doesn't seem to do anything after the observation.

jekhokie, 2021-02-28:
@royldean if you aren't already in the Discord chat, come join us here and re-post to give us as many details as you can (what you're seeing in logs, etc.): https://discord.gg/FFGDujau

There are several of us who can help walk through troubleshooting to try and figure out the issue at hand.

royldean, 2021-02-28:
Sorry, I don't do discord.    Where are the logs located?

royldean, 2021-02-28:
Hmm, found these two logs:

```
28-02-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Clearing and re-creating TLE file with latest...
28-02-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Clearing existing scheduled 'at' capture jobs...
28-02-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Scheduling new capture jobs...
INFO : Scheduling capture for: NOAA-18 20210228-205152 69
28-02-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-18 20210228-205152 69
warning: commands will be executed using /bin/sh
job 19 at Sun Feb 28 20:51:00 2021
INFO : Scheduling capture for: NOAA-19 20210228-063613 54
28-02-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-19 20210228-063613 54
warning: commands will be executed using /bin/sh
job 20 at Sun Feb 28 06:36:00 2021
INFO : Scheduling capture for: NOAA-19 20210228-180205 84
28-02-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-19 20210228-180205 84
warning: commands will be executed using /bin/sh
job 21 at Sun Feb 28 18:02:00 2021
28-02-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Done scheduling jobs!
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO : Applying any schema updates...
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 00_seed_schema.sql already applied
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 01_add_spectrogram_bool.sql already applied
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 02_add_noaa_pristine_bool.sql already applied
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 03_add_pass_azimuth_direction.sql already applied
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 04_add_capture_gain.sql already applied
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 05_add_pass_azimuth_at_max.sql already applied
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO : Schema updates complete!
28-02-2021 09:13 ./scripts/schedule.sh INFO : Clearing and re-creating TLE file with latest...
28-02-2021 09:13 ./scripts/schedule.sh INFO : Clearing existing scheduled 'at' capture jobs...
28-02-2021 09:13 ./scripts/schedule.sh INFO : Scheduling new capture jobs...
INFO : Scheduling capture for: NOAA-18 20210228-205152 69
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-18 20210228-205152 69
warning: commands will be executed using /bin/sh
job 22 at Sun Feb 28 20:51:00 2021
INFO : Scheduling capture for: NOAA-19 20210228-180205 84
28-02-2021 09:13 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-19 20210228-180205 84
warning: commands will be executed using /bin/sh
job 23 at Sun Feb 28 18:02:00 2021
28-02-2021 09:13 ./scripts/schedule.sh INFO : Done scheduling jobs!
```
```
26-02-2021 22:32 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO : Applying any schema updates...
26-02-2021 22:32 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 00_seed_schema.sql already applied
26-02-2021 22:32 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - applying 01_add_spectrogram_bool.sql
26-02-2021 22:32 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - applying 02_add_noaa_pristine_bool.sql
26-02-2021 22:32 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - applying 03_add_pass_azimuth_direction.sql
26-02-2021 22:32 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - applying 04_add_capture_gain.sql
26-02-2021 22:32 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO : Schema updates complete!
26-02-2021 22:36 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO : Applying any schema updates...
26-02-2021 22:36 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 00_seed_schema.sql already applied
26-02-2021 22:36 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 01_add_spectrogram_bool.sql already applied
26-02-2021 22:36 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 02_add_noaa_pristine_bool.sql already applied
26-02-2021 22:36 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 03_add_pass_azimuth_direction.sql already applied
26-02-2021 22:36 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 04_add_capture_gain.sql already applied
26-02-2021 22:36 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO : Schema updates complete!
26-02-2021 22:40 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO : Applying any schema updates...
26-02-2021 22:40 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 00_seed_schema.sql already applied
26-02-2021 22:40 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 01_add_spectrogram_bool.sql already applied
26-02-2021 22:40 /home/pi/raspberry-noaa-v2/db_migrations/update_database.sh INFO :   - 02_add_noaa_pristine_bool.sql already applied
[... cut ...]
27-02-2021 19:58 ./scripts/schedule.sh INFO : Clearing existing scheduled 'at' capture jobs...
27-02-2021 19:58 ./scripts/schedule.sh INFO : Scheduling new capture jobs...
INFO : Scheduling capture for: NOAA-18 20210227-210342 89
27-02-2021 19:58 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-18 20210227-210342 89
warning: commands will be executed using /bin/sh
job 18 at Sat Feb 27 21:03:00 2021
27-02-2021 19:58 ./scripts/schedule.sh INFO : Done scheduling jobs!
```

colinluthier, 2021-02-28:
It looks like it created all the images. When you say you have no images, how are you checking? The local ip of your pi should create a web panel to view them but you can navigate to them on the command line. Here are some useful file locations:

```NOAA_HOME=/home/pi/raspberry-noaa-v2
DB_FILE=/home/pi/raspberry-noaa-v2/db/panel.db
WEB_HOME=/var/www/wx-new
IMAGE_OUTPUT=/srv/images
NOAA_AUDIO_OUTPUT=/srv/audio/noaa
METEOR_AUDIO_OUTPUT=/srv/audio/meteor
RAMFS_AUDIO=/var/ramfs
NOAA_LOG=/var/log/raspberry-noaa-v2/output.log```

jekhokie, 2021-03-01:
@royldean were you able to find the images as Colin mentioned above? Thanks!

royldean, 2021-03-01:
I am using the web interface to view the images.   Again, I have two identical setups - one is working perfectly, the other one displays only a single decode from 2/28/2021.   Here is today's log file:

```
01-03-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Clearing and re-creating TLE file with latest...
01-03-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Clearing existing scheduled 'at' capture jobs...
01-03-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Scheduling new capture jobs...
INFO : Scheduling capture for: NOAA-15 20210301-080438 61
01-03-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-15 20210301-080438 61
warning: commands will be executed using /bin/sh
job 24 at Mon Mar  1 08:04:00 2021
INFO : Scheduling capture for: NOAA-18 20210301-204007 55
01-03-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-18 20210301-204007 55
warning: commands will be executed using /bin/sh
job 25 at Mon Mar  1 20:40:00 2021
INFO : Scheduling capture for: NOAA-19 20210301-175022 76
01-03-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-19 20210301-175022 76
warning: commands will be executed using /bin/sh
job 26 at Mon Mar  1 17:50:00 2021
INFO : Scheduling capture for: METEOR-M-2 20210301-082644 53
01-03-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: METEOR-M-2 20210301-08264$
warning: commands will be executed using /bin/sh
job 27 at Mon Mar  1 08:26:00 2021
01-03-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Done scheduling jobs!
```

There are no images in the /srv/images folder, and only the one decode image on the web interface (from 2/27).

jekhokie, 2021-03-02:
Hm, that's really strange. Can I suggest a re-image/re-install? Nothing in that log looks out of place.

royldean, 2021-03-02:
Yep, I agree.   It seems everything is working right, yet no images.   Currently re-imaging..... I'll report back if that works (I expect it will).

royldean, 2021-03-03:
Ok, this is getting weirder.  Pi 3B+ with a FRESH Pi OS install (burned the image this afternoon).
I sudo update/upgrade, and immediately proceeded with the raspberr-noaa-v2 install - which proceeded with no problems.   Rebooted as suggested and then went to load the web interface on my desktop computer.  Brave and Chromium browsers both show that the webpage cannot be reached (try checking the connection, try checking the proxy and firewall).  However if I open it with Firefox I get a security certificate error - but if I manually approve it - it lets me in.   No passes left today so I'll have to wait until tomorrow to see if it's getting images.... but there is the certificate error to deal with now.

jekhokie, 2021-03-03:
@royldean did you clone from a version tag, or from the master branch/latest? If the latter, the reason you're likely seeing that is because the latest changes (master branch) that will be included in the next version disable the non-TLS port 80 by default and enable port 443 with self-signed certificates in order to support password-protecting the web interface admin page.

My suggestion is one of two things:

1. Re-clone using a version tag which are "stable" releases and re-run the install.
2. 2. Update your config/settings.yml file to reflect the web settings you want. The specific section of settings that are important in this case are the following:
```
web_server_name: raspberry-noaa.localdomain
enable_non_tls: false
web_port: 80
enable_tls: true
web_tls_port: 443
cert_valid_days: 365
lock_admin_page: false
admin_username: 'admin'
admin_password: 'admin'
```

royldean, 2021-03-03:
A ha!   It works.   Only catch is I had to change "enable_non_tls:  false" to "true".

jekhokie, 2021-03-04:
Nice! Glad to hear it worked out!

closed 2021-03-04 by jekhokie