jekhokie/raspberry-noaa-v2 #193: Error after ronning install and upgrade.

tomamdy, 2021-03-01:
# pull down new code
cd $HOME/raspberry-noaa-v2/
git pull

# perform upgrade
./install_and_upgrade.sh

fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failure downloading https://github.com/wkhtmltopdf/packaging/releases/download/0.12.6-1/wkhtmltox_0.12.6-1.raspberrypi.buster_armhf.deb, ('The read operation timed out',)"}
	to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry

jekhokie, 2021-03-01:
@tomamdy I would suggest checking your network connection real quick? I just tried downloading the package from that link and it worked (meaning the link is still valid). Maybe re-try running the install script again?

tomamdy, 2021-03-01:
There must be something wrong with my Wi-fi, because with Lan he found the download direct.
Thanks.
Tom

closed 2021-03-01 by tomamdy