jekhokie/raspberry-noaa-v2 #199: Feature request - Remote access over TCP / UDP

pineapplee85, 2021-03-02:
Hi there,

I have a public QO-100 ground station using OpenWebRX and would like to add a NOAA receiver within this server (using RTL-TCP).

Would it be possible to have the SDR running as a server (Soapy or spyserver) and when the software requires it for a pass it connects to the localhost which would enable multiple clients to use the SDR at once?
This would probably require quite a lot of backend changes but thought I would put it out there as it would be very useful to me and I'm sure many others.

Thanks,
Ethan

jekhokie, 2021-03-02:
@pineapplee85 thanks for reaching out - if you're open to it, maybe join us on the Discord chat so we can flush out the details of what you're looking for? Here's a link: https://discord.gg/FFGDujau

To follow up - I've looked into Soapy and it looks pretty neat. Are you asking for whether it would be possible to have the raspberry-noaa-v2 framework instantiate a Soapy instance locally and then use it for the captures, but also yield in case there are other connected devices? I'm guessing you're looking to have the Raspberry Pi instance running the raspberry-noaa-v2 framework be a hub with Soapy, and then have the framework along with all of your other connected endpoints be the "spokes", each yielding if there is already a consumer utilizing the Soapy device?

pineapplee85, 2021-03-02:
Hi,

Yeah that sounds like what I was thinking. Would just allow multiple users on the sdr device at once and remotely. 
I'll jump on the discord now.

Cheers

jekhokie, 2021-03-04:
Closing in favor of #220 which will end up being a much more robust solution via RTMP that can then be expanded to include a more "pluggable" architecture but, more importantly, allow multiple concurrent feeds.

closed 2021-03-04 by jekhokie