jekhokie/raspberry-noaa-v2 #201: Parameter checking in test scripts

colinluthier, 2021-03-02:
A simple check for parameter passing in test scripts would be an easy upgrade. I'll try to work on it.

jekhokie, 2021-03-02:
Check out some of the other scripts already in the scripts directory - many of them already check for input parameters (number of parameters) being passed. Is there a script/set of scripts you're specifically targeting for this?

jekhokie, 2021-03-04:
@colinluthier just checking in to see if you had any other thoughts around this? Just doing my semi-usual combing through the issues list to keep things as clean as possible. :-)

colinluthier, 2021-03-05:
The one that kept getting me a has been fixed. We can close this and I will make a PR if I find others.

closed 2021-03-05 by jekhokie