jekhokie/raspberry-noaa-v2 #207: [Error] upgrade from v1 to v2, failure in script

surfrod, 2021-03-02:
**Describe the bug**
TASK [webserver : check if self-signed certificate is expired] *****************
ok: [localhost]

TASK [webserver : create self-signed certificate] ******************************
fatal: [localhost]: FAILED! => {"msg": "The task includes an option with an undefined variable. The error was: 'web_hostname' is undefined\n\nThe error appears to have been in '/home/pi/raspberry-noaa-v2/ansible/roles/webserver/tasks/main.yml': line 139, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: create self-signed certificate\n  ^ here\n"}

RUNNING HANDLER [webserver : restart nginx] ************************************
	to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry

PLAY RECAP *********************************************************************
localhost                  : ok=54   changed=38   unreachable=0    failed=1   

Error:   Something failed with the install - please inspect the logs above

**Screenshots and/or logs**
not sure what log(s) to include

**Server information (please complete the following information):**
rpi 4

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

*** just hit this, so still poking around

jekhokie, 2021-03-02:
@surfrod nice catch, I just pushed a bug fix to master - can you do a "git pull" and re-run the install script please?

surfrod, 2021-03-02:
That worked 👍 

-------------------------------------------------------------------------------
 CONGRATULATIONS!

 raspberry-noaa-v2 has been successfully installed/upgraded!

 You can view the webpanel updates by visiting the URL(s) listed in the
 'output web server url' and 'output web server tls url' play outputs above.
-------------------------------------------------------------------------------

jekhokie, 2021-03-02:
Great, happy captures!

closed 2021-03-02 by jekhokie