jekhokie/raspberry-noaa-v2 #211: create self-signed certificate

tomamdy, 2021-03-02:
Don't make the thing's comlplex.
I  get this error:

TASK [webserver : create self-signed certificate] ******************************
fatal: [localhost]: FAILED! => {"msg": "The task includes an option with an undefined variable. The error was: 'web_hostname' is undefined\n\nThe error appears to have been in '/home/pi/raspberry-noaa-v2/ansible/roles/webserver/tasks/main.yml': line 139, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: create self-signed certificate\n  ^ here\n"}

RUNNING HANDLER [webserver : restart nginx] ************************************
	to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry

PLAY RECAP *********************************************************************
localhost                  : ok=64   changed=57   unreachable=0    failed=1

jekhokie, 2021-03-02:
@tomamdy There was a bug that I fixed earlier this morning - try doing a 'git pull' and re-run the install script, it should fix this issue.

tomamdy, 2021-03-02:
Ok, it is better now.

closed 2021-03-02 by jekhokie