jekhokie/raspberry-noaa-v2 #216: Logic broken in webinterface/captures

Krolykke, 2021-03-02:
When you have several captures newest captures is on page 1 and oldest on following pages. Same logic applies to previous / next button meaning next leads to previous passes and previous to newest. It should all be opposite.

jekhokie, 2021-03-02:
@Krolykke the pagination is done in order where latest is first because I figured people would be interested in the most recent captures first - for example, if you amass thousands of captures, do you really want to page through all of the older stuff to get to the newest/latest captures?

"Next" and "Previous" buttons are actually page descriptions, not "Next Passes" or "Previous Passes" (not time-related).

What do you think? I'm still thinking it might make sense to keep it this way so when you have hundreds/thousands of captures, flipping to the Captures page gets you right to the latest first?

Krolykke, 2021-03-02:
In my head, the logic is upside down. I sort of see what you mean, but I would still suggest a change so oldest captures live on lowest numbered pages. Also, "Next" should lead to newer images and "Previous" to older. You could consider a first and last button leading to first (oldest) page and last (newest) page and/or throw a cookie that remembered where you were last. All in all, I think that a change is in place as it, at least in my head, would improve intuition :-)

Krolykke, 2021-03-09:
|-----------------------------------------------|
|    <- Older Captures  |        Newer Captures ->  |
|   <<- First page.         |               Last page ->>   |
|-----------------------------------------------|
|                     (captures listed here)                       |  <---- Could have a "Newest/oldest" first button
|                                                                                |
|-----------------------------------------------|

jekhokie, 2021-03-24:
I'm going to close this for now.

closed 2021-03-24 by jekhokie