jekhokie/raspberry-noaa-v2 #218: Add twitter posting?

surfrod, 2021-03-02:
**Please Describe the problem to be solved**
Looking to see how to enable posting of images to twitter, as v1 already does using post.py++

**(Optional): Suggest A Solution**
if it's already enabled, I can't figure out why it's not working :)

If there are multiple solutions, please present each one separately.

jekhokie, 2021-03-02:
@surfrod you *could* do this today using IFTTT connector to Twitter (the same way the Facebook IFTTT connector is described in the documentation) and enable the email handler to push to the IFTTT site. Would that suit your needs?

surfrod, 2021-03-02:
I have no idea what IFTTT is (other than the name and some uses after a cursory search just now).

I certainly don't want to involve facebook. I also don't have email enabled right now (AFAIK). The images are only being posted into /var... right?

jekhokie, 2021-03-02:
The images are stored in `/srv/images/`.

Check out this link which explains how to get email set up (https://github.com/jekhokie/raspberry-noaa-v2/blob/master/docs/emailing.md). https://ifttt.com/ has a Twitter handler you can set up that gives you an email address you would use to send images to and it would auto-post to your twitter feed, in case that's helpful?

surfrod, 2021-03-02:
I dunno.

I already have the twitter api setup, might be just as easy to edit receive** and insert :

if [ -n "$CONSUMER_KEY" ]; then
        log "Posting to Twitter" "INFO"
        if [ "${SUN_ELEV}" -gt "${SUN_MIN_ELEV}" ]; then
                python3 "${NOAA_HOME}/post.py" "$1 ${START_DATE} images: https://weather.reyni.co/detail.php?id=$pass_id" "$7" "${NOAA_OUTPUT}/images/$3-MCIR-precip.jpg" "${NOAA_OUTPUT}/images/$3-MSA-precip.jpg" "${NOAA_OUTPUT}/images/$3-HVC-precip.jpg" "${NOAA_OUTPUT}/images/$3-HVCT-precip.jpg"
        else
                python3 "${NOAA_HOME}/post.py" "$1 ${START_DATE} images: https://weather.reyni.co/detail.php?id=$pass_id" "$7" "${NOAA_OUTPUT}/images/$3-MCIR-precip.jpg" "${NOAA_OUTPUT}/images/$3-MCIR.jpg"
        fi
fi

with appropriate changes.

I don't want to wade into IFTTT unless I have to, I don't want to link images to any current email address, nor create a new one, nor worry about two factor authentication....

oh well, one way or the other...

surfrod, 2021-03-02:
Justin,

Rather than do this, I've just moved post.py over, and inserted some modified code into receive_noaa.sh.

I think I caught the env variable changes... much easier than trying to hack my email to accept IFTTT

jekhokie, 2021-03-03:
@surfrod if you post your script back to this issue in GitHub I can work it back into the framework if you have something functional?

surfrod, 2021-03-03:
Not perfect (for example I haven't subbed something for $pass_id nor fixed all the text, but I inserted this into receive_noaa.sh and it worked:

    238 # add twitter code
    239 # start of new code
    240 export NOAA_OUTPUT=/srv
    241 . "$HOME/.tweepy.conf"
    242 if [ -n "$CONSUMER_KEY" ]; then
    243         log "Posting to Twitter" "INFO"
    244         if [ "${SUN_ELEV}" -gt "${SUN_MIN_ELEV}" ]; then
    245                 python3 "${NOAA_HOME}/post.py" "$1 ${START_DATE} images: https://weather.reyni.co/detail.php?id=$pass_id" "$7" "${IMAGE_FILE_BASE}-MCIR-precip.jpg" "${IMAGE_FILE_BASE}-MSA-precip.jpg" "${IMAGE_FILE_BASE}-HVC-precip.jpg" "${IMAGE_FILE_BASE}-HVCT-precip.jpg"
    246         else
    247                 python3 "${NOAA_HOME}/post.py" "$1 ${START_DATE} images: https://weather.reyni.co/detail.php?id=$pass_id" "$7" "${IMAGE_FILE_BASE}-MCIR-precip.jpg" "${IMAGE_FILE_BASE}-MCIR.jpg"
    248         fi
    249 fi
    250 # end of new code

it depends on post.py from v1 and the 4 env variables defined in tweepy.conf

Caveat here, of course, is that some folks found getting the api access to twitter (ie getting a developer account setup) a pain.  I had no trouble doing so, therefore for me, this was an easier path to "restore" my twitter feed (found it less intrusive than trying to link up my email to IFTTT etc). However, for those who have already used IFTTT/setup etc, or have no reservations about doing so, certainly the email push that you already support should suffice and perhaps is an easier route.

surfrod, 2021-03-03:
There is a mystery here, in that I see images posted to twitter, that are not present in the Captures tab on the web portal:

e.g. the 7:14 and 7:41 passes are seen un the passes tab, and these images are posted to my twitter feed, but I do not find these images anywhere.

NOAA 15 | 07:14:31 | 07:29:34 | 51° E | 18° | Southbound
-- | -- | -- | -- | -- | --
NOAA 19 | 07:41:57 | 07:56:54 | 36° W | 5° | Southbound

I do see NOAA 18 20:28 3/2/2021 images on both the Captures tab and in my twitter feed.

Is there something about nighttime/daytime images being captured and stored (or not)?  Just discovered this mystery....

jekhokie, 2021-03-03:
Check your integration of the post script - is it possible the post is happening before the database updates in the receive script and something is failing or exiting prematurely (prior to the database being updated)?

You can also check `/var/log/raspberry-noaa-v2/output.log` for clues.

surfrod, 2021-03-04:
I don't see anything in that log, but the twitter posts reveal something wrong in my environment. The text is incorrect but the images are :)

So, I'm tracing down what is calling what right now, seems that the crontab entry was never updated to point to v2, though that doesn't explain what I'm seeing completely.

Will keep digging....

surfrod, 2021-03-05:
Ok, all is right with the world :)

Turns out I still had the v1 scheduler running in from crontab. Deleting that and rebooting then letting it run overnight has restored my captures on the portal, as well as posting to twitter. I still don’t have all the arguments passed into post.py correctly but at least the weirdness is explained and corrected.

The current inserted code in (only, for the moment) receive_noaa.sh is:

# add twitter code
# start of new code
. "$HOME/.tweepy.conf"
if [ -n "$CONSUMER_KEY" ]; then
        log "Posting to Twitter" "INFO"
        if [ "${SUN_ELEV}" -gt "${SUN_MIN_ELEV}" ]; then
                python3 "${NOAA_HOME}/post.py" "${SAT_NAME} ${START_DATE} image: ${PASS_DIRECTION}" "${PASS_DIRECTION}" "${IMAGE_FILE_BASE}-MCIR-precip.jpg" "${IMAGE_FILE_BASE}-MSA-precip.jpg" "${IMAGE_FILE_BASE}-HVC-precip.jpg" "${IMAGE_FILE_BASE}-HVCT-precip.jpg"
        else
                python3 "${NOAA_HOME}/post.py" "${SAT_NAME} ${START_DATE} image: ${PASS_DIRECTION} " "${PASS_DIRECTION}" "${IMAGE_FILE_BASE}-MCIR-precip.jpg" "${IMAGE_FILE_BASE}-MCIR.jpg"
        fi
fi
# end of new code

surfrod, 2021-03-05:
Hmm the past of the comment lines caused that formatting above :)

closed 2021-03-05 by surfrod