jekhokie/raspberry-noaa-v2 #219: Separate config sections in settings for NOAA / Meteor?

Krolykke, 2021-03-03:
_Could_ it be an idea to have separate config sections in settings for NOAA / Meteor so that one can differentiate conditions for reception on them?

E.G.:

NOAA_receiver_gain: 0
NOAA_sat_min_elevation: 25
NOAA_sun_min_elevation: -90

METEOR_receiver_gain: 15.7
METEOR_sat_min_elevation: 70
METEOR_sun_min_elevation: 10

jekhokie, 2021-03-03:
This is a great idea! I'll see if I can tackle it in the next couple of days.

jekhokie, 2021-03-10:
@Krolykke I have a PR submitted that has the configuration updates in place for everything you proposed - we'll start with this and add configurations on a per-satellite basis as people start to use it and see how people like it.

I'm going to let this "bake" on my local setup for the passes through tomorrow - if all goes well, I'll merge to master and you can pull/try it out to see how you like the functionality. It matches what you have proposed, except I did it on a per-satellite basis (not just on per satellite type basis) to allow for better/more flexibility and granularity in case people wanted to use different settings.