jekhokie/raspberry-noaa-v2 #220: "Round the World" Captures

jekhokie, 2021-03-03:
Adding a placeholder for something we discussed weeks ago related to enabling "Around the World" capability. There are 2 thoughts to explore, both of which would be "opt-in" for users of this framework:

1. Have users opt in to publish their line data to a centralized event store (Kafka or similar) that could then be used by anyone around the world to consume frequency information from other SDR devices and produce imagery to their liking. This could mean huge composite images from image frequency data that is published by various SDR devices around the world, with discovery features such as "I want data from somewhere within this lat/lon range" to help with obtaining imagery for full globe composites.
2. Have users opt in to enable remote access to their SDR device using something such as rtl_tcp or soapy and develop a client tracker that users can launch that will enable the user to track certain satellite vehicles. For example, the user could select "NOAA 19" and the software would "jump" between remote SDR devices that are online and display near-realtime imagery of what the satellite "sees", almost as though the satellite was a remotely monitored video camera.

jekhokie, 2021-03-03:
Just adding a note for myself - getting satellite current position using Predict is done via the following:

```
predict -t tmp/weather.txt -f "NOAA 15"
```

This will output (according to [this site](http://manpages.ubuntu.com/manpages/bionic/man1/predict.1.html)) the following fields:

```
 The  information  produced  includes  the  date/time in Unix format (the number of seconds
 since midnight UTC on January 1, 1970), the date/time in ASCII (UTC), the elevation of the
 satellite  in  degrees,  the  azimuth  heading of the satellite, the orbital phase (modulo
 256), the latitude and longitude of  the  satellite's  sub-satellite  point  at  the  time
 specified,  the  slant  range  to  the  satellite in kilometers with respect to the ground
 station's  location,  the  orbit  number,  and  the   spacecraft's   sunlight   visibility
 information.
```

Latitude and longitude for the satellite's sub-sat point at current time can then be used to determine a search for a receiver that is within range. If all available receivers are compiled in a database with reasonable estimates of range (lat/lon positive and negative), this information mixed with pass direction can be considered a "starting" lat/lon to search for a receiver in range. That receiver can then be picked up to record from.

dom-robinson, 2021-03-03:
oh niice - and i am pretty sure Predict is the root of hope for geotiff metadata too...

#198 etc...

pineapplee85, 2021-03-12:
Additional information: 
In my original post I suggested using something like soapy, but have just looked into spyserver [https://www.rtl-sdr.com/rtl-sdr-tutorial-setting-up-and-using-the-spyserver-remote-streaming-server-with-an-rtl-sdr/](url) and it seems a bit easier to set up. 
It also allows your receiver to be published onto the spy server map [https://airspy.com/directory/](url) with minimal config.

I would guess the base tcp server doesn't overly matter which one you use? Could be wrong though.

Hope this is of use
Ethan

pineapplee85, 2021-03-28:
Additional idea -

It would probably be beneficial to log somewhere the Signal To Noise ratio (SNR) of each path (maybe for about 15 days) to provide a 'success rate' or percentage accuracy so the software can calculate the ground station route that would probably provide the best quality captures when hopping stations.

dom-robinson, 2021-09-29:
Closing and directing to #198 since that will be the underlying tool that can allow all the maps to be merged easily and seems to be the 'indsutry standard' geo tagging for images.

closed 2021-09-29 by dom-robinson