jekhokie/raspberry-noaa-v2 #224: Automatically fill in default config parameters if not existent in

Cirromulus, 2021-03-04:
**Please Describe the problem to be solved**
When new parameters are added to the configuration file, a user has to manually copy & paste missing parameters from `config.yml.sample`. While this is very explicit, it causes more often confusion than help.

**Suggest A Solution**
When the install script detects missing parameters, ask if default values from sample config shall be copied (ideally including the comments for them).

colinluthier, 2021-03-05:
It could ask for values just as easy. 

Missing settingx: The default is true. Please enter a value or press enter to accept the default.

Krolykke, 2021-03-06:
It could also be also configured with something like accept_missing_settings_with_default: true/false/ask

True:
Missing settingx: Default setting "xxxxx" is applied, you can change it in settings.yml if you wish.

False:
Missing settingx: See config.yml.sample for details

Ask:
Missing settingx: The default is true. Please enter a value or press enter to accept the default.

closed 2021-09-29 by dom-robinson