jekhokie/raspberry-noaa-v2 #227: Target azimut?

Krolykke, 2021-03-05:
I have seen a feature in another weather sat image capture application (don't remember witch one) that let you sort of specify your "target azimut" so that if you e.g. from north over west to south are "blinded" with landcape or something else can wait for more optimal conditions before you start receiving, eventually skipping them totally. Ultimately it could be developed to a somewhat more or less "inteligent" pass analyser that based on your own input that would skip bothering with passes doomed to fail anyway.

NOAA_target_az_start: 180,0
NOAA_target_az_stop: 259,9

METEOR_target_az_start: 180.0
METEOR_target_az_stop: 259.9

colinluthier, 2021-03-06:
This can be merged with my automated cropping request. It's the same solution.

jekhokie, 2021-03-06:
Closing as the scope is included in #191

closed 2021-03-06 by jekhokie