jekhokie/raspberry-noaa-v2 #228: Polar signal plot

Krolykke, 2021-03-05:
Would it be possible to make a polar plot with your signal overlayed, so that you can visualize your signal problems in any direction?

jekhokie, 2021-03-10:
Any suggestions on how to approach this? I'm wondering if this is even possible without having an additional receiver that can obtain radiation patterns. Thoughts?

dom-robinson, 2021-03-10:
I’d love this, but afaik you need a reference signal to make it possible: https://www.rtl-sdr.com/tag/radiation-pattern/ <https://www.rtl-sdr.com/tag/radiation-pattern/> 
-- 
Dom Robinson
uk.linkedin.com/in/domrobinson

Krolykke, 2021-03-11:
Sure, I agree, no problem. -just a thought on your comment about second receiver, in my head, I don't see why you would need that because the signal is recorded and saved (depending on configuration). At least with NOAA, there is generated a nice timeline with a spectrogram where you see the carrier and the signal. I imagine that some sort of representation of that could be generated on a polar plot following the satellites path.

jekhokie, 2021-03-11:
That would be interesting in particular - I understood this to originally be related to the RF signal, but if you're talking about satellite pass on a polar plot, I think that would be pretty cool as an output and even potentially overlaid on the image as "telemetry".

closed 2021-03-12 by jekhokie