jekhokie/raspberry-noaa-v2 #230: No Image Captures

rto54, 2021-03-06:
I, sorry, but i tried everything, but, i still con not receive images regardless of the stelittes, help

My configuration is a fallows
- Rpy 3 B +
- SSD
- Nooelec NESDR SMArt v4 SDR
- QFH antenna
- Nooelec SAWbird + NOAA Barebones

[image]
[image]
[output.log](https://github.com/jekhokie/raspberry-noaa-v2/files/6095721/output.log)

```
NOAA_HOME=/home/pi/raspberry-noaa-v2
DB_FILE=/home/pi/raspberry-noaa-v2/db/panel.db
WEB_HOME=/var/www/wx-new
IMAGE_OUTPUT=/srv/images
NOAA_AUDIO_OUTPUT=/srv/audio/noaa
METEOR_AUDIO_OUTPUT=/srv/audio/meteor
RAMFS_AUDIO=/var/ramfs
NOAA_LOG=/var/log/raspberry-noaa-v2/output.log
LAT=48.625117
LON=6.297899
SCHEDULE_NOAA=true
SCHEDULE_METEOR=true
SAT_MIN_ELEV=0
SUN_MIN_ELEV=0
DELETE_AUDIO=false
BIAS_TEE="-T"
SDR_DEVICE_ID=0
FLIP_METEOR_IMG=true
GAIN=0
TZ_OFFSET=1
LOG_LEVEL=DEBUG
NOAA_MAP_CROSSHAIR_ENABLE=true
NOAA_MAP_CROSSHAIR_COLOR=0xffff00
NOAA_MAP_GRID_DEGREES=10.0
NOAA_MAP_GRID_COLOR=0xff0000
NOAA_MAP_COUNTRY_BORDER_ENABLE=true
NOAA_MAP_COUNTRY_BORDER_COLOR=0xffff00
NOAA_MAP_STATE_BORDER_ENABLE=true
NOAA_MAP_STATE_BORDER_COLOR=0xffff00
PRODUCE_SPECTROGRAM=true
PRODUCE_NOAA_PRISTINE=true
PRUNE_OLDER_THAN=0
PRUNE_OLDEST=0
METEOR_RECEIVER=rtl_fm
FREQ_OFFSET=-6
NOAA_CROP_TELEMETRY=true
IMAGE_ANNOTATION_LOCATION=NorthEast
GROUND_STATION_LOCATION='Saint Nicolas de Port - France'
NOAA_DAY_ENHANCEMENTS='ZA MCIR MCIR-precip MSA MSA-precip HVC-precip HVCT-precip HVC HVCT therm'
NOAA_NIGHT_ENHANCEMENTS='ZA MCIR MCIR-precip therm'
ENABLE_EMAIL_PUSH=false
EMAIL_PUSH_ADDRESS=test@ifttt.com
ENABLE_EMAIL_SCHEDULE_PUSH=false
ENABLE_DISCORD_PUSH=false
DISCORD_WEBHOOK=''
NTP_SERVER=ntp.midway.ovh
WEBPANEL_PORT=80
ENABLE_SATVIS=true
```

Krolykke, 2021-03-06:
Come join the chat on [Discord](https://discord.gg/A9w68pqBuc), there are a bunch of nice guys that certainly will help solve your issue :-)

rto54, 2021-03-06:
`I Krolykke RTO54 is KAribou54 on discord

Krolykke, 2021-03-07:
Issues solved on Discord and can be deleted/closed @jekhokie

closed 2021-03-07 by jekhokie