jekhokie/raspberry-noaa-v2 #231: Strange interference patterns or otherwise a bad reception

tomamdy, 2021-03-06:
I have done this part and find the CSV file, but I do something wrong and dont can get the heatmap waterfall image.
I think that I do somethig wrong. Please give me a suggestion.

If you experience strange interference patterns or otherwise a bad reception, it may come from strong FM Radio stations,
PC switchmode powersupplies, or nearly anything else. If you want to analyze this, you can run these scripts overnight to see
whats floating around.
```bash
cd $HOME
raspberry-noaa-v2/scripts/testing/scan_for.sh 5h
```
This will scan a range for five hours and produce a heatmap waterfall image in your current working directory afterwards.
The output of the script will show the file that will be created.

Krolykke, 2021-03-07:
Come join the chat on [Discord](https://discord.gg/A9w68pqBuc), there are a bunch of nice guys that certainly will help solve your issue :-)

tomamdy, 2021-03-07:
I don't like Discord. For me Rubbish.

closed 2021-03-07 by tomamdy