jekhokie/raspberry-noaa-v2 #232: SNR/Gain optimization

Krolykke, 2021-03-07:
**Please Describe the problem to be solved**
A script to help in the process of finding optimal SNR by adjusting gain on signal on a specified frequency.

**(Optional): Suggest A Solution**
Optimizing SNR can in theory be done in an easy 3 step process:

1. Reduce the gain to the lowest setting.
2. Find a signal near 137-145 that's stable, perferably not from a moving satellite.
3. Slowly raise the gain UNTIL the signal level STOPS RISING. As soon as the SIGNAL stops increasing STOP adjusting the gain.

So, I suggest a new script in ./raspberry-noaa-v2/scripts/testing that can do that.

The script would for the time take 2 parameters, namely frequency and auto/manual.

Then it could flow like this:

gain_optimize 138 auto

1. Examine and remember the supported gain values of SDR and set it to its lowest value above zero.
2. Tune to the specified frequency
3. Read (out) the signal level and register it
4. Increase gain to next supported value, wait for the signal to stabilize again
5. Has signal level increased? Repeat from 3
6. Print out optimal gain that would be the one from before, just before the signal level stoped rising. 

If you give the parameter "manual" the script should run in a manual mode where you by pressing +/- adjust gain and get the signal level and the specified frequency read out.

When needed, another parameter could be implemented to specify what SDR to use if more is attached.

Some sort of error/stupidity handling should probably be implemented as well.

jekhokie, 2021-03-12:
@Cirromulus this seems right in your wheelhouse given your experience with the excellent scan tooling you already provided. Care to take a crack at this?

Cirromulus, 2021-03-12:
I indeed have a "benchmark" script that decodes a continuus AM-Transmission of an Airport nearby at ~133MHz.
But this only got me into the rough ballpark, and I won't recommend it, because:

1.  Narrow-Band AM is pretty different to our FM-like NOAA APT, so "optimal" sound may deviate strongly between them
2. All "nearby" signals have inherently different amplitudes that the Sats we are trying to receive, so you may end up optimizing for a much stronger signal than you would have
3. I did not calculate SNR in a script yet, because this would have to be analyzed from I/Q data, not the already demodulated signal.

My easy, but manual proceeding was:

1. Disconnect SDR from PI and connect the "pipeline" to a laptop
2. Pop up SDR# or GQRX during a pass
3. Manually adjusting gain, LNA position and antenna orientation while checking live with the calculated SNR.
4. Note findings and configure r-noaa settings.

jekhokie, 2021-03-24:
Closing as a bit out of scope/extremely difficult to imagine how this could be done - we can open this back up when there is a bit more time to focus on this type of SDR-based tuning.

closed 2021-03-24 by jekhokie