jekhokie/raspberry-noaa-v2 #235: Ability to Overlay Pass Elevation on Images

jekhokie, 2021-03-08:
It would be neat to add a vertical axis to the images produced that would indicate satellite elevation over the period of the capture. This would help with setting min elevation for different use cases.

For example - using Predict, we can get the elevation of the satellite at different intervals. Using the image height of the captured image (in pixels), we could normalize the total pass duration over this and line up the elevation at various epoch times reported by predict on the image itself at the y-pixel locations.

Krolykke, 2021-03-11:
-could you plot the satellites path over the picture as well? IMHO it all should take place in a separate "picture enhancement".

Ultimately wet dream: the path should indicate signal quality/strength on the line, perhaps small tags with numbers along it.

dom-robinson, 2021-03-11:
+1 nice idea :)
-- 
Dom Robinson
uk.linkedin.com/in/domrobinson

colinluthier, 2021-04-09:
The path, by definition is the center of the image. Having its vector in degrees would be an easy add. It's simply the equation for the angle created by a line between two points in polar notation described by AOS and LOS.

dom-robinson, 2021-09-29:
is this practicable? Still open?

colinluthier, 2021-09-29:
This is pretty easy. It would put a scale on one side of the image showing elevation. It needs some information back from wxtoimg about auto-cropping.

colinluthier, 2022-01-28:
I am going to close this. After consideration there is no easy way for me to figure out where the autocrop has happened with out decoding the telemetry bars before and after the crop.

closed 2022-01-28 by colinluthier