jekhokie/raspberry-noaa-v2 #241: Choice of satellites

rto54, 2021-03-09:
Hello, is it possible to be able to choose the satellites that we wish to receive or not to receive.
Another question, why don't we have the METEOR MN2-2?
 Thanks

Krolykke, 2021-03-09:
Come join the chat on [Discord](https://discord.gg/A9w68pqBuc), there are a bunch of nice guys that certainly will try to answer your questions ;-) We would be so happe to see you there again.

rto54, 2021-03-09:
Ok, I thought my requests were requests for evolutions ;-)

jekhokie, 2021-03-10:
Circling back on this - Meteor MN2-2 is only transmitting HRPT which requires a dish antenna with tracker and quite a bit of processing power, so it's out of scope at the moment for this framework, which is optimized for the Raspberry Pi.

I can, however, see a need to select which satellites you wish to track - that's actually something that funnels nicely into "per-satellite settings" that we've been discussing, so I'll leave that part of the feature request here for consideration.

closed 2021-03-10 by jekhokie