jekhokie/raspberry-noaa-v2 #245: Auto Twitter post images

PretorianJ, 2021-03-10:
Auto email pushing is not very clear how to set it up. 
Did the configuration but it doesn't seem to work. Anyone who did it and worked? 

i will be using IFTTT for auto tweetting

dom-robinson, 2021-03-10:
Ha - that’s was one of mine :)

So basically create a free account on IFTTT.

Add the ‘EMAIL’ function:

.

This will require an email you want to use to send triggers from (i setup a gmail specifically for this) - ONLY messages sent FROM that email TO trigger@applet.ifttt.com <mailto:trigger@applet.ifttt.com> will ‘kick the trigger’.

That forms the ‘If This then’ part of the ‘recipe’.

The ’Then That’ is up to you. You can trigger anything IFTTT supports. 

In my case i set up a facebook page, and connected it to IFTTT specifically and so all my emailed photos post on the facebook page.

I hope that helps :) 

Dom

PretorianJ, 2021-03-10:
> Ha - that’s was one of mine :)
> 
> So basically create a free account on IFTTT.
> 
> Add the ‘EMAIL’ function:
> 
> 
> This will require an email you want to use to send triggers from (i setup a gmail specifically for this) - ONLY messages sent FROM that email TO trigger@applet.ifttt.com <mailto:trigger@applet.ifttt.com> will ‘kick the trigger’.
> 
> That forms the ‘If This then’ part of the ‘recipe’.
> 
> 
> The ’Then That’ is up to you. You can trigger anything IFTTT supports. 
> 
> In my case i set up a facebook page, and connected it to IFTTT specifically and so all my emailed photos post on the facebook page.
> 
> 
> 
> 
> I hope that helps :) 
> 
> Dom
> 
> > 
> > 
> > Auto email pushing is not very clear how to set it up.
> > Did the configuration but it doesn't seem to work. Anyone who did it and worked?
> > 
> > i will be using IFTTT for auto tweetting

Yes I did the IFTTT part and i have the email. My issue is with the push email on the Rpi side. E.g trying to run mpack for test and it does not. Also one question on the gmail ID section in the msmtp file what did you put?

jekhokie, 2021-03-10:
@PretorianJ for the mail section, the parameters are the following:

```
account: gmail
host: smtp.gmail.com
port: 587
from: <your_email_address>@gmail.com
user: <your_login_user_id>
password: <your_login_password>
```

If you have 2FA/MFA enabled on your gmail account, however, you'll need to get an application password and put that into the "password" field above. If you don't, simply use the password you typically use to log into your gmail.

Also, if you're having issues, try checking out the logs in `/var/log/raspberry-noaa-v2/output.log` to see if there is anything that stands out as problematic.

Lastly - for troubleshooting, it's much easier over chat - come join us on Discord, there are many of us there working on this framework and helping users get on their way! https://discord.gg/NywJEPP5

PretorianJ, 2021-03-10:
Awesome. Will Join then once i go Home as i am at work atm. Ok on the email ID section what do you put? Also how do i know when the push is triggered?

jekhokie, 2021-03-10:
@PretorianJ just checking if you were able to get this resolved from our discussion on Discord?

PretorianJ, 2021-03-11:
Close This :) We resolved it :) 

Just informationally for the parameters remove <> and it should work :)

closed 2021-03-11 by PretorianJ