jekhokie/raspberry-noaa-v2 #247: Ground Station Notation

colinluthier, 2021-03-11:
If the ground station is left blank delete the text "ground station" from the annotation.

jekhokie, 2021-03-11:
@colinluthier I'm assuming you mean the annotation in the push scripts, correct? Because the annotation in the images is customizable by the user now (if you set it, it's there - if you remove it from the annotation template, it's not).

The former is an issue that I can see, but it's bigger than just ground station - we need to address the following:

* ground_station_location
* show_sun_elevation
* show_pass_direction

Since those are all configurable as true/false, the correct thing needs to happen in the `scripts/receive_noaa.sh` and `scripts/receive_meteor.sh` scripts (components are included/excluded from the posted annotation based on what the user selects in the configuration for true/false).

I'll mark this as a bug, albeit a bit wider in scope than the original issue filed - nice find.  👍🏻

jekhokie, 2021-03-11:
On second thought, I realize now this is likely a regression - when we published one of the previous versions (v1.4.0) we removed these parameters prematurely as they are still needed for the non-image annotations (e.g. email push, discord push, etc.). I'm going to need to re-add these to the configurations and add conditional logic to handle displaying/hiding these parameters.

jekhokie, 2021-03-11:
@colinluthier circling back on this - can you clarify real quick for me? Which annotation are you referring to - I see that in the push notifications the ground station is excluded if the text is a blank string? If you're referring to the image annotation, you need to update your `config/annotation/annotation.html.j2` template to remove that because that's the customized way that users now modify their annotation.

colinluthier, 2021-03-11:
I don’t remember who’s, but yesterday one of the discord pushes had the
image annotation as “Ground station:” I am not sure if it was a
configuration problem but wanted to note it in case it was a bug.

jekhokie, 2021-03-11:
I'm going to close this for now - I triple checked the logic and not to say I didn't miss anything, but I can't see how this would happen in the latest version of the code base.

closed 2021-03-11 by jekhokie